prusa3d/PrusaSlicer #3502: large bedsize causes PrusaSlicer to crash

JBA-flex, 2020-01-09:
### Version
PrusaSlicer-2.1.1+win64-201912101512

_Use `About->About PrusaSlicer` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
Win 7 and win 8

### Behavior
When setting bedsize to x:3000 Y:2000 PrusaSlicer crashes without any warning or error messages. 
x:2000 Y:2000 works without any issues.

bubnikv, 2020-01-09:
It is a known issue, for performance reasons the coord_t type (fixed point type rounded to 1/10 of a micrometer) is a 32bit int.

bubnikv, 2020-03-25:
Duplicate of https://github.com/prusa3d/PrusaSlicer/issues/3833
and https://github.com/prusa3d/PrusaSlicer/issues/2208

closing this one.

closed 2020-03-25 by bubnikv