prusa3d/PrusaSlicer #3503: [Feature request] Advanced Modifiers - First Layer Width

DjjC13, 2020-01-09:
2.1.1 Win64

Windows 10 1909

MKS 1.2 & MKS 1.3 & MKS 1.4 - Marlin 1.1.9 & Marlin 2.0.x bugfix

It would be nice to have options within the "add modifiers" to allow the user to tweak the advanced / extrusion width options more.

First layer extrusion width is especially important to me.

I routinely create a few modifier meshes that intersect models. I then will create a sparse infill for the "core" of the component, A medium infill for the "mantle", and then a shell for the skin.

To reduce material, and make the back of the part look especially "engineered" I like to set the thin infill sections to have 0 bottom layers, this creates a honeycomb effect on the back side of the part

Problem is that I would like to "over extrude" the sections that are the thin infill for the first layer to get a better bite on the built platform. Effectively purposely exaggerating the elephants foot phenomenon to get better adherence.

As seen in the picture (viewed from beneath), it would be nice to have the option to set the "First Layer" to say 175% for the red honeycomb infill, then have the purple, yellow, and orange sections work from the "nominal" first layer properties.

Until now I have been either doing this manually on the printer by adjusting the flowrate, or in the gcode to do the same effect. Its tedious and requires a gcode simulator to know where to insert the codes.

[image]

DaleKramer, 2022-03-12:
I too would like to be able to have access to 1st layer extrusion width for each part on the platen.

My reason is different from the OP's in that some thin wall parts on the platen print better with smaller 1st layer extrusion width.

Here is an example:
[image]

[image]

mepoland, 2023-01-17:
+1 for being able to control 1st layer extrusion width for individual parts within an object. I like to create multicolored inlays in my 1st layers, as shown below. To get a clean inlay, I want to eliminate the 2-perimeter portions (see image).
[image]

I can achieve this by increasing 1st layer extrusion width in Print Settings. But this affects the whole print, leaving me with unnecessarily fat & ugly infill in the rest of the part (see 2nd image).
[image]

Currently there is no way to change this with modifiers. There is an "external perimeters" that is useless on the 1st layer because it is overridden by overall Print Settings' 1st Layer Extrusion Width.
[image]

closed 2026-09-29 by prusaslicer-bot[bot]