prusa3d/PrusaSlicer #3505: Place on face has to be used twice

espr14, 2020-01-09:
v2.2.0 alpha2 win64

I would like to place the model upside down so I use place on face function. First application causes rotation but wrong face of the rectangle is used. After second application on the same face causes the model to be placed properly so it can be printed.

[lever.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/4042412/lever.3mf.zip)

[image]

lukasmatena, 2020-01-10:
Thanks, you are correct. Fixing the model in NettFabb did not help either. We will look into it.

lukasmatena, 2020-10-29:
This issue should be fixed with 1130d78c19ffca26ea21632a98640be4f7053a6d for PrusaSlicer 2.3.0. Closing.

closed 2020-10-29 by lukasmatena