prusa3d/PrusaSlicer #3506: Supports generated outside of enforcer volume

GarthSnyder, 2020-01-10:
### Version
2.2.0-alpha2+

### Operating system type + version
macOS Catalina 10.15.2

### 3D printer brand / version + firmware version (if known)
Sidewinder X1

### Behavior
This is likely pilot error of some sort. Nevertheless... 

In the attached file, a support enforcer slab hugs the X and Y boundaries of the printable object but extends beyond the object vertically. "Generate support material" is turned on, but "Auto generated supports" is turned off. (The help for "Auto generated supports" says, "If unchecked, supports will be generated inside the 'Support Enforcer' volumes only.")

However, on slicing, supports are generated outside the enforcer volume. The extraneous supports are _related_ to geometry within the enforcer volume, but they do not themselves fall within the volume.

Just to be clear, my intention was to limit supports to being bars within the vertical gap only. There should be no supports on the side of the object and no support network on the build plate.

[image]
[image]
[image]
[SupportMischief.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/4044108/SupportMischief.3mf.zip)

_Is this a new feature request?_
No

bubnikv, 2021-09-22:
This should be fixed with the new snug supports in prusaSlicer 2.4.0-alpha1

closed 2021-09-22 by bubnikv

GarthSnyder, 2021-09-22:
Great news!