prusa3d/PrusaSlicer #3513: Please implement gradient infill

daktylus, 2020-01-11:
### Feature Request

Please implement a gradient infill as shown from Stephan form CNC Kitchen.

Everything is well explained in his video: [https://www.youtube.com/watch?v=hq53gsYREHU]

thenickdude, 2020-01-12:
I've implemented a patch to support gradient infill in my PrusaSlicer fork here:

https://github.com/thenickdude/PrusaSlicer/commit/c0e4f006f6a6277df2f14ec09d343fe25a647890

https://www.reddit.com/r/prusa3d/comments/enjx4d/i_added_support_for_cnc_kitchens_gradient_infill/

nordurljosahvida, 2020-03-19:
Related but on the z axis: https://github.com/prusa3d/PrusaSlicer/issues/3770

Stephan also mentions cura's gradual infill in the video.

Tupson444, 2026-09-13:
still wanted