prusa3d/PrusaSlicer #3514: feature request/suggestion - remove skirts from __default__ profiles

Mirarkitty, 2020-01-11:
### Version
any

#### Suggestion

Remove skirts from any default print profile. They don't fill any function on a Prusa MK3 with a normal print. They are remnants from earlier printers.

Printing a skirt will risk the skirt getting lose and ruining the print. Not printing with skirt makes very little difference in almost any print.

The purge line already fills half the function of a skirt and the default skirt is not making a difference in cooling of the bottom layer.

bubnikv, 2020-01-14:
Thanks for suggestion. We believe the skirt has its purpose to verify the 1st layer thickness before starting the print.

Closing.

closed 2020-01-14 by bubnikv

Mirarkitty, 2020-01-14:
That's a third explanation, one I never heard before. But shouldn't it make sense to limit it to one layer then?

lukasmatena, 2020-01-14:
I believe it is three layers so it is easier to remove it.

bubnikv, 2020-01-15:
> I believe it is three layers so it is easier to remove it.

Exactly. With the structured PEI sheets or even the smooth PEI sheets, the
single layer skirt may be very difficult to remove.