prusa3d/PrusaSlicer #3515: Mouse Cursor freezing within Prusa Slicer and when moving out of or back into Prusa Slicer Window

smrogers91, 2020-01-11:
### Version
_Version of PrusaSlicer used goes here_

PrusaSlicer V2.2.0Alpha2+Win64

_Use `About->About PrusaSlicer` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #s_
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_
[image]

### 3D printer brand / version + firmware version (if known)
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_

Prusa I3 Mk2s - Presets added only for Prusa Mk2s 0.4mm and 0.25mm nozzle as well as filaments noted in image below.
[image]

### Behavior
* _Describe the problem_
* _Steps needed to reproduce the problem_
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*PrusaSlicer*__ preview are preferred_

I have a dual monitor setup. When the cursor moves in and out of the Prusa Slicer window the mouse will freeze for 1-3 seconds before resuming movement. The problem also occurs occasionally when only moving the mouse within the Prusa Slicer window and not moving the mouse outside of the frame.

When first observed, CPU, Memory, and GPU, and both hard drives were under less than 10% load, and Google Chrome was the only other application open. 

Problem resolved when Google Chrome (Version 79.0.3945.117 (Official Build) (64-bit)) was closed and began again when Google Chrome was reopened.

The problem does not persist with Prusa Slicer being open in the background and moving the cursor around otherwise. The problem does not persist when Prusa Slicer is closed. 

_Is this a new feature request?_

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_

zalexzperez, 2020-02-17:
I'm having the same problem...it is driving me crazy.
PS 2.2.0 alpha-4 here.

bubnikv, 2020-02-18:
what is the make of your computer? What is the graphics card model? How old
is the computer? Is it a new model?
Are those monitors 4k? (high resolution)
Do they have a different font scaling resolution set?

smrogers91, 2020-02-18:
Bubnikv, I'm assuming you were asking Zalexzperez?

This problem appears to have resolved with Alpha4

zalexzperez, 2020-02-18:
I lied, it's not the same problem.
The mouse pointer just gets stuck every few seconds for no apparent reason on Alpha-4.
I switched back to 2.1.0RC and the problem is gone.

My laptop is a Ideapad700 running Windows 10, with an external 24" 1080p monitor configured as extended display. Intel i7 6700HQ and GTX950m graphics card.

luckyguido, 2020-02-24:
i think i have the same issues as zalexzperez, i'm on a desktop intel i5 with nvidia 970 using wireless mouse and keyboard, prusa slicer 2.2 alpha4 and the new 2.2 beta has the issue .
If you keep moving mouse over slicer window cause it freeze for a second, doing the same over any other window mouse work correctly. 
Now the funny thing, it happens only with a Trust keyboard, with another, a logitech one, it work perfectly. Any other program (or for example version 2.1) work like a charm.
I hope this can anyway help to find a solution.

trust keyboard model 16597-02
logitech keyboard model MK220

bubnikv, 2020-02-28:
We have implemented two new features:
Detecting the 3DConnextion mouse plug / unplug events.
Detecting an external media (SD card, flash drive) plug / unplug events.

Both these detections run at idle of the main user interface message queue processing and they are executed with 2 second intervals. It may be that one of these tests are blocking possibly due to a faulty hardware driver.

We plan to move these two tests in to a worker thread, so they will not block the user interface, possibly to 2.2.0-rc, certainly to 2.2.0 final. We are also considering to use operating system support for notifications on disk drive & USB device changes, but that is OS dependent and considerably more involved than polling.

I have one question regarding the mouse hanging. Does the mouse cursor hang, or is it "just" that the PrusaSlicer UI becomes unresponsive? This is an important question, because on Windows the mouse cursor is painted by the operating system, so it should move even in case the PrusaSlicer's main thread is blocked due to polling the disk drive change or 3DConnexion plug / unplug events.

smrogers91, 2020-02-28:
Bubnikv, When I was having the problem in Alpha2 the cursor would hang for a second. It was impossible to tell if PrusaSlicer was unresponsive. The cursor would hang essentially every time I moved the cursor outside of the UI.

luckyguido, 2020-02-28:
mouse completely stop moving. 
is it possibile to easily disable these features you mentioned at compile time? 
I can try to make some test disabling one feature at time.

bubnikv, 2020-02-29:
@luckyguido

> is it possibile to easily disable these features you mentioned at compile time?
> I can try to make some test disabling one feature at time.

That would help.

Polling of the removable medias to show / hide the eject button:
GUI_App.cpp:277
#if !__APPLE__
		RemovableDriveManager::get_instance().update(wxGetLocalTime(), true);
#endif

Polling of the 3DConnextion mouse plug / unplug is done in
GLCanvas3D.cpp:2666
Replace
    bool mouse3d_controller_applied = wxGetApp().plater()->get_mouse3d_controller().apply(m_camera);
with 
 bool mouse3d_controller_applied = false;

Thanks.

luckyguido, 2020-03-03:
@bubnikv i'm having some trouble compiling deps, i'v got this error:
`6>C:\Users\guido\source\repos\PrusaSlicer\deps\build\dep_openexr-prefix\src\dep_openexr\IlmBase\Half\half.cpp(58,13): error C2059: errore di sintassi: 'const' [C:\Users\guido\source\repos\PrusaSlicer\deps\build\dep_o
       penexr-prefix\src\dep_openexr-build\IlmBase\Half\Half.vcxproj] [C:\Users\guido\source\repos\PrusaSlicer\deps\build\dep_openexr.vcxproj]
     6>C:\Users\guido\source\repos\PrusaSlicer\deps\build\dep_openexr-prefix\src\dep_openexr-build\IlmBase\Half\eLut.h(6,1): error C2143: errore di sintassi: ';' mancante prima di '{' [C:\Users\guido\source\repos\PrusaSli
       cer\deps\build\dep_openexr-prefix\src\dep_openexr-build\IlmBase\Half\Half.vcxproj] [C:\Users\guido\source\repos\PrusaSlicer\deps\build\dep_openexr.vcxproj]`

sorry for late answer

bubnikv, 2020-03-07:
We have reworked two services that were introduced in PrusaSlicer 2.2.0-alpha: Detection of 3DConnexion USB plug / unplug events and removable USB driver or SD card plug / unplug events. These detection tasks were performed at a user interface thread, while now they are executed in their respective background threads in PrusaSlicer 2.2.0-rc2. Plase let us know whether this refactoring resolved your issues. We are closing this issue. Please feel free to reopen if your issue has not been resolved with PrusaSlicer 2.2.0-rc2.

luckyguido, 2020-03-12:
I tested with RC3 and on another PC using the same keyboard, issue is still present, what is changed is that on previous versions the mouse was freezing only when the pointer was over the application window, now it freeze everywhere in the screen when slicer is open. I hope this can help

bubnikv, 2020-03-12:
@luckyguido So it happens to you with trust keyboard model 16597-02
It looks like your USB enumeration driver for that particular device is bad, or the device itself blocks badly when enumerating.

@smrogers91 Is the issue resolved for you with PrusaSlicer 2.2.0-rc3?

@zalexzperez It happens with Victsing wireless keyboard and mouse to you. Is the issue resolved for you with PrusaSlicer 2.2.0-rc3?

bubnikv, 2020-03-12:
> @luckyguido So it happens to you with trust keyboard model 16597-02
> It looks like your USB enumeration driver for that particular device is bad, or the device itself blocks badly when enumerating.

I mean, it looks like your whole windows are hanging when enumerating that device.

bubnikv, 2020-03-12:
You are guys all on Windows 10?

bubnikv, 2020-03-13:
Would you please test with this build?

Now on Windows both the 3DConnexion USB HID (Human interface device) enumeration and disk enumeration runs only when notified by the operation system about device plug / unplug events. This should hopefully fix the mouse stutter issue you are experiencing with some other USB devices plugged in.

The removable drives are still polled by this build, but only once per 30 seconds. That is to detect if a USB flash disk was ejected using file explorer or the windows UI, which event is not reported to applications by Windows (two of our developers spent considerable time with that).

bubnikv, 2020-03-15:
Let's hope this issue will be fixed with PrusaSlicer 2.2.0-rc4.
Closing. Please feel free to reopen if it is still an issue.

closed 2020-03-15 by bubnikv

luckyguido, 2020-03-16:
@bubnikv rc4 seems to be OK!

zalexzperez, 2020-03-16:
Yes, RC4 is also working fine. My Victsing wireless keyboard and mouse are having no issues now.

bubnikv, 2020-03-16:
@luckyguido @zalexzperez finally a victory! Enjoy.