prusa3d/PrusaSlicer #3516: Skirt prints on top of brim

thsturgill, 2020-01-12:
### Version
_2.2.0 Alpha2
_Use `About->About PrusaSlicer` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_Linux, Fedora 
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_

### 3D printer brand / version + firmware version (if known)
_Prusa Mini, Prusa MK2.5S
### Behavior
* _Describe the problem_
* _Steps needed to reproduce the problem_
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*PrusaSlicer*__ preview are preferred_
Unless brim "distance from object" is manually set to a value greater than the size of the brim, the skirt will print on top of the brim instead of outside it. 
_Is this a new feature request?_
NO

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_

BeldrothTheGold, 2020-01-13:
Hi. This was a requested feature added to slicer in 2.2.0 alpha 1. 

From release notes:

*The skirt is newly allowed to overlap with the brim, so that the skirt, when used as a wind draft shield is stabilized by the brim. Brim lines are split at the intersections so there is no overextrusion #724.

salmander, 2020-01-14:
Same on osx build [see here](https://photos.app.goo.gl/YEF6y9MKzSSMn5rj7)

lukasmatena, 2020-01-14:
@salmander Could you post a 3MF? Thanks.

BeldrothTheGold, 2020-01-14:
@lukasmatena it seems like most folks think this is a bug. Maybe consider making this an advanced option, disabled by default? Allowing skirt overlap on brim that is.

bubnikv, 2020-01-14:
I don't think it is a good idea to add another set of switches. IMHO it is better to document the current behavior. Our content guys are working diligently on first version of PrusaSlicer manual.

salmander, 2020-01-14:
> @salmander Could you post a 3MF? Thanks.

Hey, I, unfortunately, can't as it is not an open-source model.  Is there any other way I can help?

thsturgill, 2020-01-14:
I would agree with bubnikv - Once I knew what was happening it is no problem to change the position if I want. It was more that we were not expecting this behavior change.

lukasmatena, 2020-01-14:
@BeldrothTheGold Apart from the fact that we generally want to avoid needless parameters and switches, in this case it even makes no sense. When you set skirt height and skirt distance from object, you have uniquely decided where the skirt should be. So the checkbox would have to say "ignore something you've already set and instead do something to the skirt so it does not cross the brim, because... I like it". @thsturgill basically complains that setting skirt distance from object to 5 places the skirt 5 units from the object. It should have always been this way and it was fixed because:
1. people wanted control over distance from the object
2. people wanted skirt to connect to brim, so it sticks to the bed.
Both of these reasons make sense. I think the behaviour is correct.

@salmander Yes - make sure you actually see a bug and not the intended behaviour. I bet it is in fact correct, but cannot tell for sure from that picture alone.

BeldrothTheGold, 2020-01-16:
> @BeldrothTheGold Apart from the fact that we generally want to avoid needless parameters and switches, in this case it even makes no sense. When you set skirt height and skirt distance from object, [...]

> I don't think it is a good idea to add another set of switches. IMHO it is better to document the current behavior. Our content guys are working diligently on first version of PrusaSlicer manual.

I apprecaite the well thought out responses.  I was, foolishly, trying to offer a suggestion to avoid additional issues. Clearnly not a sustainable way to write software. (You can't make everyone happy)

I know the first time I saw the current behavior I also thought it might be a bug. I was even planning to create an issue until I saw there was already an open issue. It wasn't until I read @bubnikv's response on the other issue that I understood it was the intended behavior. (and documented as such in the release notes of alpha 2) 

I do agree the current brim/skirt behavior is more flexible and I do agree that education is the answer to avoid additional issues. Thanks for all your hard work on PrusaSlicer!

lukasmatena, 2020-01-16:
@BeldrothTheGold Thanks. I'll close this issue then. If there is an actual bug in some scenario, it can be reopened.

closed 2020-01-16 by lukasmatena