prusa3d/PrusaSlicer #3518: Feature request: preserve top layer thickness with auto layer height

fpikus, 2020-01-12:
### Version
2.2.0 alpha

### Operating system type + version
Any

### 3D printer brand / version + firmware version (if known)
Prusa MK2/3

### Behavior
When the automatic variable layer height is used, it's really hard to set top layer count correctly, which results in severe pillowing sometimes (when all top layers are at minimum height or close). A better option would be to preserve the top layer thickness and determine the count from the layer height at the corresponding Z.

_Is this a new feature request?_
Yes

#### Project File (.3MF) where problem occurs
_

Area5142, 2020-01-14:
See issue #3411

lukasmatena, 2020-02-06:
Closing. See #3411 for details.

closed 2020-02-06 by lukasmatena