prusa3d/PrusaSlicer #3521: Rotating Generic Box about Z axis actually rotates about Y

Incendium87, 2020-01-14:
### Version
2.1.0

### Operating system type + version
Windows 7

### 3D printer brand / version + firmware version (if known)
I3 Mk3
### Behavior
* _Describe the problem_
Modifier boxes import already rotated 90° about X. trying to rotate about Z rotates about Y instead. The arrows that show up to show direction of rotation show proper about Z but the actual motion is about Y.

* _Steps needed to reproduce the problem_
Import modifier box and rotate about axis. Directions appear to be miss-coded.
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
Modifier box to rotate about Z axis corresponding to the Z axis of the printer or Z axis of the modifier.

* _Actual Results_
Box rotated about printer/object Y

  * _Screenshots from __*PrusaSlicer*__ preview are preferred_
[image]

_Is this a new feature request?_
No

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_

[Prusa Slicer Rotation.zip](https://github.com/prusa3d/PrusaSlicer/files/4057616/Prusa.Slicer.Rotation.zip)

neophyl, 2020-01-14:
I think its because the axis are fixed relative to the enforcer.  Its more apparent whats happening if you rotate your enforcer 45 degrees around X first and then try and rotate around Z, The Z (and other axis) moves with the part rather than staying locked to the World space.

Wether that's a bug or intended design operation I don't know but both methods are certainly valid ways of handling it.

Incendium87, 2020-01-16:
I agree that the axes move with the enforcer, and I think that this behavior would be fine and perfectly logical if it was consistent. My initial problem was that when the enforcer was first created it was rotated about X by 90° already. Then, when I wanted to rotate about Z, which was the long axis of the enforcer (the 30 mm edge), the enforcer did not rotate about the long axis even though the arrows displayed on screen seemed to show that it would.

An additional and probably related problem I have just found is that the rotation, axes orientation, and scaling of the enforcer is not persistent and resets to an odd orientation. When I reopened my .3mf file to respond to your comment, I found the rotation, scaling, and size had been reset. All rotations were 0° and all scaling was at 100%. This caused the originally rotated part to have dimensions of 28.28x28.28x10 (xyz) rather than the original 10*10*30. With dimensions of 28.28x28.28x10 and with the enforcer in the orientation shown, this indicates that the coordinate system that the enforcer defaulted to has Y up (along machine Z) and Z to the front of the machine (along machine y). This is confirmed by the way the arrows highlight on screen when you click into the size text field to change the size of the enforcer.

[image]

Streeter1981, 2020-04-12:
I'm finding that the rotation is just completely off on any modifiers --- the axes do not seem to align with anything in particular. 

I just added a modifier and here are the default axes available for rotation, which are just nonsensical are REALLY not useful:
[image]

Due to this issue, I may have to downgrade back to 2.1 as I simply cannot use any modifiers because I can't put them in any position. By default the rotation is set to the below values when I add the modifier:
[image]

Again that's _before_ I've even touched the modifier.

Erwos, 2021-01-02:
I think my problem is related to this one. When I move the Enforcer box in the Y axis, the Z value changes in the Part Manipulation Position box instead of the the Y. Makes me wonder if the Y and Z got swapped.

[image]

closed 2026-09-29 by prusaslicer-bot[bot]