prusa3d/PrusaSlicer #3524: Derivated settings on height range modifiers gives bugs

Tillema, 2020-01-14:
### Version
PrusaSlicer 2.1.1. Alpha 2

### Operating system type + version
Windows 10

### 3D printer brand / version + firmware version (if known)
Original Prusa i3 MK2S

### Behavior
When I add a few extra perimeters within height range, the infill interferes with the perimeters at that extra perimeters.
In order to make it better printable I want to give the first few layers extra dense (by setting infill density to 50%), but that doesn't do it's job, so thought it the other way around by adding extra perimeters, but then this problem occurred.

#### Project File (.3MF) where problem occurs
[Knoop2.zip](https://github.com/prusa3d/PrusaSlicer/files/4059345/Knoop2.zip)

lukasmatena, 2020-01-14:
I do not see anything wrong. You can

1. Try to reproduce it in PrusaSlicer2.2.0-alpha2
2. Take a screenshot of the problem and post it here
3. Save the faulty project as 3MF (not AMF) and post it here

Thanks.
Could be somehow related to #3382, but who knows.

lukasmatena, 2020-02-11:
Is this still an issue?

bubnikv, 2020-03-11:
no response, closing.

closed 2020-03-11 by bubnikv