prusa3d/PrusaSlicer #3526: No dark theme.

NotErikUden, 2020-01-14:
_Is this a new feature request?_
Yes

The PrusaSlicer has a critical flaw built in its machinations. There seems to be no dark theme set as the standard theme from the beginning, even worse, there seems to be no dark theme at all. I hereby sincerely request a dark mode as an addition to the otherwise wonderful slicer. 

Regards

lukasmatena, 2020-01-14:
You might consider reading [release log](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.2.0-alpha1) of latest alpha. Also, it is customary to report application version and your platform when filing bug reports of "critical flaws".

Does this answer the issue?

bubnikv, 2020-02-19:
no response, closing.

closed 2020-02-19 by bubnikv

phazei, 2020-10-29:
Any update on this?  Has MS released the Dark Mode API for Win32 controls?  From what I can find it's only available for UWP API's, so waiting for that could be years.  Is there no way to just add theming so people could manually choose dark mode?

https://stackoverflow.com/questions/51334674/how-to-detect-windows-10-light-dark-mode-in-win32-application

lukasmatena, 2020-10-30:
The discussion shall continue in #4493 (if there is something left to discuss). This one is a duplicate and it is closed, please do not comment on it.