prusa3d/PrusaSlicer #3529: Skirt clips into second object of a two-part STL

thedevleon, 2020-01-16:
### Version
PrusaSlicer-2.2.0-alpha2+win64

### Operating system type + version
Windows 10

### 3D printer brand / version + firmware version (if known)
Heavily modified Ender 3 with Duet Wifi.

### Behavior
I'm trying to print the [heatbed cable cover](https://github.com/prusa3d/Original-Prusa-i3/blob/MK2/Printed-Parts/stl/heatbed_cable_cover.stl) for the MK3. However, after rotating the part fo that it faces onto the build plate, the skirt around it is only created for one of the two pieces, and clips into the second part. The first layer of the second part is completely missing.

[image]

[image]

[image]

#### Project File (.3MF) where problem occurs
[heatbed_cable_cover.zip](https://github.com/prusa3d/PrusaSlicer/files/4068624/heatbed_cable_cover.zip)

neophyl, 2020-01-16:
The issue is that the second part is actually placed spatially higher than the first and so the bottom of part 2 (the smaller part) is not placed on the bed. If you split to Objects and then use the place on bed tool on part 2 it will then slice correctly and the first layer will be on the bed and the skirt will correctly go around both parts.

[image]

FidelCapo, 2020-01-16:
This is duplicate of https://github.com/prusa3d/PrusaSlicer/issues/927
Closing.

closed 2020-01-16 by FidelCapo