prusa3d/PrusaSlicer #3530: Don't keep Bed heated if last unload failed

redlukas, 2020-01-16:
### Version
2.2.1

### Operating system type + version
Windows 10

### 3D printer brand / version + firmware version (if known)
Hardware: i3 MK3S & MMU2S
Firmware: MK3S: 3.8.1, MMU2S 1.0.6

### Behavior
Observed Behaviour:
If the last unload of the Filament fails, the Printer goes into "MMU needs attention" mode. This lets the Nozzle cool down, but keeps the Bed warm
Expected Behaviour:
The "cooldown Bed" would be sent before the last "unload filament" command. In this case, it would still display the "MMU Needs Attention" error, but let the heated bed cool down
Advantages if implemented:

- Less danger of the hot bed (fire/burn risk)

- Less power consumption

- If you are lazy like me and use a short bowden tube and a small piece of filament to trick the FINDA when printing in MMU Single mode to avoid the tedious process of loading the filament trough the Buffer the plate will stay hot, even if the print is otherwise finished and this will let the bed cool down.

_Is this a new feature request?_
Yes

###Related Issue
[Firmware Issue #2419](https://github.com/prusa3d/Prusa-Firmware/issues/2419)

lukasmatena, 2020-01-16:
I don't think this is PrusaSlicer related issue and should be discussed (and IMHO closed) in the FW issue tracker that you referenced. Closing here.

closed 2020-01-16 by lukasmatena

Panayiotis-git, 2020-01-16:
I don't find this proposition is a good idea. If we let the bed to cool, then it is the most possible that the printed part will be detached from the bed.

redlukas, 2020-01-16:
> I don't find this proposition is a good idea. If we let the bed to cool, then it is the most possible that the printed part will be detached from the bed.

What would be the issue if the part detached from the bed? why do you want to avoid this?

NovaViper, 2020-01-17:
Some beds have glass, so if it cools down, then the part will come off the bed, hence resulting in a complete reprint of the object.

redlukas, 2020-01-17:
yeah, i get that.
but what i am talking about is the LAST unload. This happens after the part is completely printed, as part of the "finishing up"(you know, things like lifting the head and moving it away from the object, letting the nozzle cool etc) routine. Since the print is already completely done, i dont see the harm if the part detaches from the build plate.

NovaViper, 2020-01-17:
Ah, OK. Then I would think that's something particularly in the firmware to mess around in, not in the slicer itself. Though as how to do this, I have no clue since I don't own a Prusa nor a MMU myself. 

⁣Get BlueMail for Android ​

redlukas, 2020-01-17:
Well, i believe the issue should be tackled primarily in the slicer and the firmware second:

- The slicer determines the order in which this "finishing up"-script is written in the GCODE. This could be changed without much hassle

- the firmware could look ahead in the GCODE and see that there is no more extrusion being done. therefore, it is safe to turn off the bed heating.

NovaViper, 2020-01-17:
The main problem with the first part is that the slicer has *no way* of knowing rather the MMU failed to unload or load, only the firmware is able to figure that out. The slicer basically goes and just slices up the object you feed it; it's up the firmware to be able to plan for the 'unexpected'. When the firmware does find that it can't unload, then it should take over and turn off everything (just like how thermal runaway violations are handled)

redlukas, 2020-01-19:
yeah, i am not saying the slicer should detect the failed unload. i am saying the standard order of the end script should be changed. with the proposed change, the M104 S0 (Turn off heatbed) command would appear in the GCODE sooner than the G1 E-15.0000 F3000 (unload filament) gcode.

wpegden, 2020-01-29:
The slicer should simply add a command in the gcode to unset the bed temperature before the last unload.