prusa3d/PrusaSlicer #3533: The wipe tower adds an extrusion line in mid air on the final layer

Auge103, 2020-01-17:
**### Version**
2.2.0-alpha2+win64 
[220e2db]
(Bug happens in 2.1 release as well)

**### Operating system type + version**
Windows 10 x64 Professional [Version 10.0.17763.973]

**### 3D printer brand / version + firmware version (if known)**
Prusa MK3s + MMU2s, both default.

**### Behavior**
**Description:**
When trying to slice a multi material object with manually added extruder changes only in the first 10 layers, the generated wipe tower adds adds a floating extrusion (in mid air) on the final layer.

[image]

**What should happen:**
The wipe tower should end after the last tool change.
**What actually happens:** 
The wipe tower ends after the last tool change, then adds an extrusion line in mid air at the final layer.

**#### Project File (.3MF) where problem occurs**
[Case-Main-SlicerBug.zip](https://github.com/prusa3d/PrusaSlicer/files/4077993/Case-Main-SlicerBug.zip)

(There are a few extrusions on the support structure that are in mid air as well)

jmd73, 2020-01-17:
Please read the FAQ and search the issues before reporting. This has been reported many times and is not a but.
This is from the FAQ:
Why there is an extrusion hanging in the air over a multi material wipe tower?

That is the final ramming before the filament is pulled from the nozzle and parked into the filament switcher. See #1355 for further details.

closed 2020-01-18 by Auge103

tbrummel, 2020-01-21:
@jmd73 @lukasmatena 
Ok, I seeing that this 'issue' has been reported a lot -- and you're both telling people it's by design.

I do understand it's the final ramming to get melted filament out of the hot zone -- However, what I'm not seeing explained is why it's printing in mid air as opposed to on the wipe tower?  Does this extrusion have to happen in mid air?  The fact that's it too high in z space is what I consider the 'issue' - not the fact that it's happening.  (and if it doesn't need to happen in space, it seems lukasmatena's latest changes would make it easy to 'fix' in this release as he's tracking wipe tower z)

BTW, lukasmatena, I've been running the latest beta with your changes to support what I call an  'intelligent wipe tower' ("sparse" I think it's called in the app) and I LOVE IT!  I have printed many multi-color jobs with the beta and have had no problems with the feature.  It saves so much filament!  I'm glad to see that the MMU2 survey comments were read and taken to heart (or maybe it was already on the list -- I'm just happy to see it)  I think a lot of people are going to consider it the best feature of the next release - if they find and understand it).  Cheers!

lukasmatena, 2020-01-21:
@tbrummel Thanks for the compliments and the testing. You are right that the low wipe tower is more useful than I assumed it would be. And it will be even more useful given the changes that @YuSanka has done on the ColorPrint feature (although there are still some changes coming).

As for the final extrusion in mid-air. If you're suggesting to lower z and reach the top wipe tower level, that opens space for collisions with the print. And we don't want that, because it is one more think the user can forget about and it also makes the usable area of the bed smaller. I think that would not stop complains from coming. Current behaviour is IMHO better, the "issue" is at least harmless.

tbrummel, 2020-01-21:
Ok, so it would be nice if it were printed on the tower -- but clearly printing the tower to the end z height of the object is a bad idea and I agree that most users would not want to deal with tower placement issues. 
 
So, for now, we live with it -- but a couple projects could be added to the list:
1) If using the new "sparse" mode,  we could print it on the tower as the user is already ensuring no collision with printed objects.  That would be nice and I imagine relatively simple right now given your code changes. 

2) In the future, PrusaSlicer could be updated with code to detect collisions by adding an "extruder size Width, Height and Offset X and Y" to define the rectangle from the nozzle tip that defines the print head size (a polygon would be overkill I think).  Since printer definitions are already part of the Slicer, defaults for the Prusa printer models would be easy to build in and, of course, put it on an advanced page for custom printers.  Then the slicer could warn that the tower and the print head will collide and force the user to change placement (and, better yet, just use this info in auto-arrange so there is no problem).  Agreed that this does reduce the build area a bit but for most prints, this is not a problem.  I don't see much value in adding this in if users are not using your new feature for Sparse towers -- but I think the sparse towers should be the default mode of operation once automatic collision detection is programmed in.  Then we've got the best of everything - low waste, no collision management for users, and no more posts about printing into thin air :-)

Do you know who updates the FAQ?  I'd like to get the fact that collision prevention is why it prints into thin air added to the answer.