prusa3d/PrusaSlicer #3534: [FEATURE REQUEST] Please add additional toolchange variables.

vintagepc, 2020-01-18:
### Version
2.2.0-alpha

### Operating system type + version
OpenSuSE 42.3

### 3D printer brand / version + firmware version (if known)
Mk3S/MMU2S

_Is this a new feature request?_
Yes

### Feature desired:

Please expose more variables in the custom toolchange gcode area. This will allow for easier integration of custom toolchange scripts that still respect filament settings. I realize that not everything can be exposed, but I'm looking for a few specific ones that cannot be parsed out of the gcode settings section or easily calculated. Specifically:

- toolchange number (not available at all if no wipe tower). Very handy for reference/comment-gcode.
- Wipe volume that must be purged for this toolchange (while accounting for wipe space in objects and infill).

These would probably be tricky but nice to have if possible:
- Ramming sequence for current filament
- cooling move sequence for current filament
- previous and next filament properties

Much appreciated!

vintagepc, 2020-01-18:
Edit - I didn't quite RTFM that some of those are already available but not as convenient, e.g. the correct tool must be used to reference filament properties etc. 

But convenience references for those would be superb. 
I'm going to keep digging to see which other ones are already exposed.

bynnek, 2020-01-27:
I'd also like to see this as it would help with custom purge mechanisms.

bubnikv, 2020-01-30:
I believe this is working already.

Just use the previous_extruder and next_extruder indices in your
toolchange_gcode.
For example
`{filament_ramming_parameters[ previous_extruder]}`
shall work.

bubnikv, 2020-01-30:
I believe this is working already.

Just use the previous_extruder and next_extruder indices in your
toolchange_gcode.
For example
```{filament_ramming_parameters[ previous_extruder]}```
shall work.

vintagepc, 2020-01-30:
Oh, cool! I will have to take a look for that! (though my inspection of the code did reveal wipe volumes were not available , hence the PR for that - the extras I mentioned are nice but not critical.)

vintagepc, 2020-02-01:
I just reviewed it and realized I wasn't very clear. I do see the raw parameters are available and can be used to recreate the sequence. What I meant to ask for was the ramming and cooling gcode sequence. But, in retrospect that was a silly request, as it is tied to the wipe tower X/Y moves and is not very usable without actually having a wipe tower at that X/Y. I updated the original request.

closed 2026-09-29 by prusaslicer-bot[bot]