prusa3d/PrusaSlicer #3535: [Feature Request] Paint multiple colors on single stl files

MimbaMonkeyHouse, 2020-01-18:
**Version**
2.2

**Operating system type + version**
All OS

**3D printer brand / version + firmware version (if known)**
MK3S (FW v3.7.0) with MMU2S (FW v1.0.5)

**Behavior**
Add the possibility to paint models directly in PrusaSlicer, the way Mosaic Canvas does. I really want to use PrusaSlicer for creating multicolor models, rather than rely on Meshmixer... Right now, I need to use an alternative slicer.

rgojr2001, 2020-01-18:
All for it!

MimbaMonkeyHouse, 2020-01-18:
For users not familiar with what I mean, have a look here:
https://www.bigbrain3d.com/product/simplycolor3d-software-to-make-any-3d-print-multicolor/

MimbaMonkeyHouse, 2020-01-18:
Or here:
https://www.mosaicmfg.com/blogs/news/updated-and-completely-refreshed-canvas-now-lets-you-digitally-paint-your-models

smplfy, 2020-01-18:
Please! This would be amazing!

batmangb, 2020-01-18:
Yes please! 👍🏻

shro70, 2020-01-18:
Yeah great idea. We need really  this feature

Ducati87, 2020-01-18:
Great feature! Hope to see it implemented into PS.

TheRealGuldukat, 2020-01-18:
This is a "MUST" for MMU...

gizzburn, 2020-01-18:
Sounds like a great addition request.

grajjio, 2020-01-19:
Oh yes please!

jbbe1962, 2020-01-19:
Would love to have all my 3D needs in Prusa Slic3r

noahdiamond4, 2020-01-20:
Yes!!!! This would convince me to but a mmu!

carpenda, 2020-01-21:
I'm all in for this feature.  Prusa needs to provide this to keep up with competition.

carpenda, 2020-01-21:
I just saw a reference to Pepr3D, designed with the help of Prusa, and used to paint single color stl files for importing into PrusaSlicer.

MimbaMonkeyHouse, 2020-01-21:
Turns out there is such a software that PRUSA Research contributed to. Josef mentioned it at a Q&A session at a States-side convention. Video is here: https://www.youtube.com/watch?v=jh2tMyOjG1I
This app's v1.0 is open-source, compatible with PRUSA and works relatively well in spite of its beta feel. It's available on GitHub here: https://github.com/tomasiser/pepr3d/releases/download/v1.0/Pepr3D-Windows-x64.zip

MimbaMonkeyHouse, 2020-01-21:
Do we need a 3d painter integrated with PrusaSlicer or is a standalone compatible and opn-source app just as good?
/polls Integrated 'Standalone'

MimbaMonkeyHouse, 2020-01-21:
> I just saw a reference to Pepr3D, designed with the help of Prusa, and used to paint single color stl files for importing into PrusaSlicer.

Yup, see the links above...

mlegloanec, 2020-04-11:
Please! This would be amazing for my MMU2S 😃

scooper4711, 2020-07-02:
Now that we can "paint" support enforcers and blockers, surely this feature can be too far off?

tabiznet, 2020-10-08:
Need this desperately, for some reason meshmixer does not play nice with my models

3D-Land, 2021-02-28:
My idea and question would be the following, in mmu, could it be possible to paint the objects with a brush and choose the selected parts to print with a certain color?
As the brush for supports was implemented, please consider the option, separating the object into several parts is for design professionals, I only have basic knowledge

bubnikv, 2021-06-03:
Implemented in PrusaSlicer 2.4.0-alpha1

closed 2021-06-03 by bubnikv

gordo3di, 2021-08-17:
The painting thing is awesome. I did notice it crashes out if I don't exit painting mode before slicing and manually painting on the model (rather than bucket fill) is incredibly laggy. I usually have to wait 3-5 seconds before I can see the model painted but it's understood it's alpha.

jmd73, 2021-08-17:
@gordo3di Of background slicing is in, it is very laggy. For me it works Good of background slicing is off.

gordo3di, 2021-08-18:
Weird. I have background processing off. The first time I was using the program (with an instance of 2,3 open as well) everything was very laggy especially manipulating the view but most of that went away when I restarted so it may just be something different on my end.