prusa3d/PrusaSlicer #3537: PrusaSlicer has encountered an error while slicing

electrickery, 2020-01-18:
### Version
PrusaSlicer 2.1.1+ linux-64 (AppImage)

### Operating system type + version
Xubuntu 18.04 LTS

### 3D printer brand / version + firmware version (if known)
i3 MK3S + MMU2S (not used to produce the problem, so firmware version is not very relevant (3.8.1-2869 & 1.0.6-372).

### Behavior
* Importing the *.stl or *.3mf occurrs without problem. 
* Slicing fails with an PrusaSlicer error (see image), and the slice rendering is corrupted in an interesting way. Alas not-printable.
[image]

The __same__ curvedHolder.stl files was sliced (and printed) succesfully a year ago with Slic3r (then current version). Now the old .stl as well as a new one (generated with OpenSCAD 2019.05) and the equivalent .3mf give the same result in PrusaSlicer. To me it looks like a regression.

_Is this a new feature request?_
No

#### Project File (.3MF) where problem occurs
[curvedHolder2.zip](https://github.com/prusa3d/PrusaSlicer/files/4080761/curvedHolder2.zip)

If all went well, there is a PNG screenshot attached as well as a ZIP with a SCAD and 3MF file.

BirkBinnard, 2020-01-18:
I'm running Ver. 2.2.0-Alpha2 on Win10/64. I got the same results that you showed with the 3MF file. Then I used 3D Builder to save it as an STL file. Again, got same results. Then I loaded the STL file into 3D Builder; it said there were errors, so I repaired them and saved it as STL again. That version sliced perfectly.

My conclusion is there is something not right with the 3MF file, but I don't have a piece of software that will tell me what the problem is.

electrickery, 2020-01-18:
Hi BirkBinnard,
Thanks for the response and checking.

The original STL (attached) was succesfully sliced a year ago with Slic3r and I just uploaded it to some online viewers/repair services (https://www.formware.co/onlinestlrepair, https://www.viewstl.com/, https://beta.sharecad.org/en/) and no problems were found. Next I'll try installing Slic3r again and see what happens.

[curved_1.stl.zip](https://github.com/prusa3d/PrusaSlicer/files/4081605/curved_1.stl.zip)

BirkBinnard, 2020-01-18:
Well this is an odd one for sure. I opened the original STL file in one of my viewers, and it looked fine. I then opened it in Rhino, and it looked fine. I checked for naked edges, and there weren't any.
So whatever it is that's wrong with the file is pretty obscure. There may be some software out there that does a better job of checking for errors, or at least identifying what they might be, but I don't know about anything more than what I have. 
Maybe one of the gurus at Prusa can figure it out.

FidelCapo, 2021-02-01:
This issue should be fixed in next release.
Closing.

closed 2021-02-01 by FidelCapo