prusa3d/PrusaSlicer #3538: Option "Don't support bridge" not work for 2.1.1 and 2.2.0 alpha1 and alpha2 versions.

Shama71, 2020-01-18:
Option "Don't support bridge" not work for 2.1.1 and above versions.

bubnikv, 2020-03-15:
please provide models, screenshots etc.

Shama71, 2020-03-15:
[image]
This problem also in 2.2.0-rc4
[Y carriage - motor.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/4334298/Y.carriage.-.motor.3mf.zip)

bubnikv, 2020-03-15:
I see a thin gap fill in the center of that bridge. Maybe that is why that
part of the object is still supported.

Shama71, 2020-03-15:
The same object but a bridge on the other side.
[image]

rtyr, 2020-04-27:
@Shama71 Please provide 3MF project file.

Shama71, 2020-04-27:
[Y carriage - motor.zip](https://github.com/prusa3d/PrusaSlicer/files/4541976/Y.carriage.-.motor.zip)

bubnikv, 2021-02-24:
For my reference, I believe the gap fill is guilty here

[image]

[image]

bubnikv, 2021-02-24:
The other bridge seems to work fine, no supports were generated.

[image]

closed 2026-09-29 by prusaslicer-bot[bot]