prusa3d/PrusaSlicer #3539: "unable to enumerate HID device" error and crash when slicing

fitzroy99, 2020-01-18:
### Version
_Version of PrusaSlicer used goes here_

_Use `About->About PrusaSlicer` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #s_
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_

### 3D printer brand / version + firmware version (if known)
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_

### Behavior
* _Describe the problem_
* _Steps needed to reproduce the problem_
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*PrusaSlicer*__ preview are preferred_

_Is this a new feature request?_

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_

fitzroy99, 2020-01-18:
Version 2.2.0 alpha 2 when slicing for new Mini.
When slicing for 3rd party printer - no problem.

fitzroy99, 2020-03-04:
Crashes while slicing on all slicer versions and all .stl files for Mini. Works fine for Printrbot 1405 Makers.

bubnikv, 2020-03-05:
> Version 2.2.0 alpha 2 when slicing for new Mini.

Does it work with PrusaSlicer 2.2.1?

> When slicing for 3rd party printer - no problem.

You mean when slicing with PrusaSlicer 2.2.0 for 3rd party printer it is not a problem?

fitzroy99, 2020-03-05:
Yes.

bubnikv, 2020-03-06:
yes what?

fitzroy99, 2020-03-06:
Yes slicer works for a 3rd party printer. All versions of Prusaslicer act
the same. Crashes all .stl files while slicing for Mini. Get "unable to
enumerate....." message each time.

fitzroy99, 2020-03-06:
Still not working for Mini but does work for i3 MK3 but shows same error
message.

bubnikv, 2020-03-07:
> All versions of Prusaslicer act the same.

All PrusaSlicer 2.2.0 versions? Or all versions of PrusaSlicer? Please try to be as explicit as possible.

> Crashes all .stl files while slicing for Mini. Get "unable to enumerate....." message each time.

Where do you get this message displayed? Could you please make a screenshot?

PrusaSlicer 2.2.0-rc2 will be released today. Would you please retest with PrusaSlicer 2.2.0-rc2? We have reworked the 3DConnexion support and the message "unable to enumerate" seems to stem from there.

fitzroy99, 2020-03-07:
Problem occurs on all versions of Prusaslicer, including 2.2.0-rc2. Does 
work for i3 MK3. Screenshot of error included.

fitzroy99, 2020-03-07:
Acts the same on all versions of Prusaslicer, including 2.2.0-rc2, but 
works for i3 MK3. Screenshot of error included.

fitzroy99, 2020-03-07:
re CRASHES

bubnikv, 2020-03-09:
What is the operating system you are running PrusaSlicer on?
If you are on Windows, are you running a 32bit or a 64bit PrusaSlicer build?
How did you install PrusaSlicer? From the drivers package or downloaded from github releases, or did you compiled PrusaSlicer yourself from the source code?

bubnikv, 2020-03-09:
You did not manage to attach the screenshot.

fitzroy99, 2020-03-09:
Windows 7 Home Edition 64 bit.
Downloaded from GotHub site.

bubnikv, 2020-03-10:
I tested PrusaSlicer 2.2.0-rc2 on an old Windows 7 enterprise 32bit laptop with old graphics card, therefore I had to use the MESA software renderer. I did not encounter any issue with the Mini presets.

JohnnyDeer, 2020-03-10:
Hi @fitzroy99 , your screenshots are not loaded, can you try it again please?

fitzroy99, 2020-03-11:
I deleted everything Prusaslicer related , installed 2.2.0-rc3. Same 
results. Works for i3 MK3 but not for Mini. Same error message.

bubnikv, 2020-03-12:
Could you make a screenshot of that troubled message you are talking about?
Would you please make a video?

fitzroy99, 2020-03-16:
I have sent screen shot several times and have asked for confimation of receipt. No answer. My new Mini arrives in 2 days and I have no working slicer. I would like some advice as to how to proceed.

bubnikv, 2020-03-16:
I confirm that no screenshot has yet been attached to this issue by you. It would be great if you can drag & drop an image to the github issue page.

bubnikv, 2020-03-16:
This is the page where the data provided by you is collected. You may verify yourself that neither screenshots nor videos passed throught.

https://github.com/prusa3d/PrusaSlicer/issues/3539

bubnikv, 2020-03-16:
You may try to contact support. Hopefully they are better trained than me on how to collect data from a customer. Please refer to this github issue when talking to support.

fitzroy99, 2020-03-16:
2700895794 Heteis a screen shot. Please let me know if you get this.

bubnikv, 2020-03-16:
We did not receive any screenshot as you can easily verify yourself by opening the following link:
https://github.com/prusa3d/PrusaSlicer/issues/3539

fitzroy99, 2020-03-27:
Have installed new Win 7, tried Slicer 2.2.0, both 32 and 64 bit, same result-crashes when slicing for Mini but works for I3. Error 0xc0000005? I now have my new Mini working well, but no slicer. Must be something that can be done.

fitzroy99, 2020-04-15:
Why do all slicer versions work for i3 printers but not for mini? Always crashes while slicing. Same for all stl files. Works for 3rd party printer.

fitzroy99, 2020-04-17:
Anybody there?

bubnikv, 2020-04-20:
I have passed your complaint to our support team.

Misfi, 2020-04-21:
Hi @fitzroy99
Would you be so kind as to upload the previously mentioned screenshots to https://imgur.com/upload? 
Make sure as much as possible is visible - do not crop the screen unless it's absolutely necessary.

You can then link the screenshots here, either as a full gallery or separate links to each one.

bubnikv, 2020-10-09:
No response, closing.

closed 2020-10-09 by bubnikv