prusa3d/PrusaSlicer #3543: Right clicking, to add a pause on macOS Catalina causes a scroll

SimplicityGuy, 2020-01-19:
### Version
2.2.0-alpha2+

### Operating system type + version
macOS Catalina 10.15.2

### 3D printer brand / version + firmware version (if known)
Prusa MK2s latest version of firmware

### Behavior
After slicing a model, when trying to add a pause at a certain layer using right-click in the layer view causes the layer view to scroll.

#### Project File (.3MF) where problem occurs
This happens with any project.

YuSanka, 2020-01-21:
Thanks for your report. Unfortunately, it's a known bug on OSX : right click event causes a little scrolling. Like a workaround you can use Ctrl+LeftMouseClick instead of RightMouseClick. And we will add this information to the tooltip on OSX, till this problem wouldn't be resolved.

SimplicityGuy, 2020-01-22:
Thanks! I'll give that a try.

SimplicityGuy, 2020-01-23:
> Thanks for your report. Unfortunately, it's a known bug on OSX : right click event causes a little scrolling. Like a workaround you can use Ctrl+LeftMouseClick instead of RightMouseClick. And we will add this information to the tooltip on OSX, till this problem wouldn't be resolved.

This suggestion works consistently. So this is a good way to get unblocked.

closed 2026-09-29 by prusaslicer-bot[bot]