prusa3d/PrusaSlicer #3547: Build issue - missing dependency for Cereal library

KyleMaas, 2020-01-19:
### Version
version_2.2.0-alpha2-110-g8840cb090

### Operating system type + version
Gentoo Linux

### 3D printer brand / version + firmware version (if known)
Irrelevant to this.  Build issue.

### Behavior
Trying to build from source results in a compilation error:

fatal error: cereal/access.hpp: No such file or directory

Installing Cereal system-wide fixes this, but it should probably be either caught at the cmake level before the build starts or included with the source.

#### Project File (.3MF) where problem occurs
Irrelevant since this is a build issue.

lukasmatena, 2020-01-19:
Have you followed the [instructions](https://github.com/prusa3d/PrusaSlicer/blob/master/doc/How%20to%20build%20-%20Linux%20et%20al.md) and compiled the dependencies beforehand?

KyleMaas, 2020-01-19:
Using cmake you mean?  Yes.  Those instructions are what I was using to build it.  And cmake did not catch the missing dependency for Cereal.  It did catch the one for nlopt, which I installed, but since it didn't catch this I figured it'd be worth reporting.

schiffsratte, 2020-01-20:
Same here - cmake warns about boost and wxpython version, catches missing nlopt but fails to catch cereal. make stops with the error KyleMaas described

lukasmatena, 2020-01-20:
The problem is that you don't have cereal headers. Whether it manifests in cmake or during compilation would not make much difference to me. Cereal is header only and all that cmake does is that it adds its include directory. I don't know if it should check for its existence (and if it can do so easily).

Maybe bundling the (small header-only) library into PrusaSlicer source tree would make more sense. I don't know if there is a reason it is done this way. But it does not qualify as "build issue" to me personally.

Anyway, instead of relying on whatever your system does or does not provide, it is possible to download and build the dependencies yourself and then tell PrusaSlicer to use them (statically). This process is automated and described in the instructions I linked to. If you do it that way, you should have no issues with neither nlopt, cereal, Boost versions etc.

KyleMaas, 2020-01-20:
Up to you how you want to handle it.  I didn't want to static link everything, which is why I didn't do that.  I'm on Gentoo Linux - if I'm building a program, I'd rather have it use system libraries.

Either way, it works on my system now that the dependencies are installed, so this bug no longer impacts me.  I just would have expected that cmake would have complained about cereal like it did with nlopt.  But it's a goofy bug that if you're running a system that has Cereal installed before you start, you'd never find, so I figured it'd be worth reporting.

Bundling it into the PrusaSlicer source tree seems like a sane solution to this.  Or adding a check into the cmake configuration like there is for nlopt.  In my view, either one would solve this.

zougloub, 2020-01-20:
@KyleMaas you would be happy with something like:
```patch
 # Find the Cereal serialization library
+INCLUDE (CheckIncludeFiles)
+CHECK_INCLUDE_FILES (cereal/cereal.hpp HAVE_CEREAL_H LANGUAGE CXX)
+if (NOT HAVE_CEREAL_H)
+  message(FATAL_ERROR "Cereal not found, please install (http://uscilab.github.io/cereal/)")
+endif()
 add_library(cereal INTERFACE)
 target_include_directories(cereal INTERFACE include)
```
... right?

KyleMaas, 2020-01-20:
Can't say I know cmake enough to know if that'd function correctly, but from the looks of it, yeah.

closed 2020-02-07 by tamasmeszaros

KyleMaas, 2020-02-07:
Excellent.  Thank you!

commit 2fb1acd7 (2020-02-07): Add check for cereal library in CMake

Fixes #3547, replaces #3613