prusa3d/PrusaSlicer #3549: Broken builds on big-endian architectures

hyperair, 2020-01-20:
### Version

2.1.1

### Operating system type + version

Debian Sid

### Behavior

Fails to build on big-endian architectures:
```
/<<PKGBUILDDIR>>/src/admesh/stl_io.cpp: In function ‘bool stl_write_binary(stl_file*, const char*, const char*)’:
/<<PKGBUILDDIR>>/src/admesh/stl_io.cpp:155:35: error: no match for ‘operator+’ (operand types are ‘std::vector<stl_facet>’ and ‘size_t’ {aka ‘long unsigned int’})
  155 |   memcpy(buffer, stl->facet_start + i, 50);
      |                  ~~~~~~~~~~~~~~~~ ^ ~
      |                       |             |
      |                       |             size_t {aka long unsigned int}
      |                       std::vector<stl_facet>
```

Full build log here: https://buildd.debian.org/status/fetch.php?pkg=slic3r-prusa&arch=ppc64&ver=2.1.1%2Bdfsg-1&stamp=1579393754&raw=0

lukasmatena, 2020-01-20:
Duplicate of #2879, already fixed with commit 94212fa.

hyperair, 2020-01-20:
Oh yeah you're right, thanks

closed 2020-01-20 by hyperair