prusa3d/PrusaSlicer #3551: curl: (1) Protocol "`https" not supported or disabled in libcurl

ggzengel, 2020-01-20:
### Version
PrusaSlicer-2.2.0-alpha2+linux-x64-201912221342.AppImage

### Operating system type + version

#lsb_release -a
```
No LSB modules are available.
Distributor ID:	LinuxMint
Description:	Linux Mint 19.3 Tricia
Release:	19.3
Codename:	tricia
```

### Behavior

After starting Slicer I get this error on console:
```
Error getting: `https://files.prusa3d.com/wp-content/uploads/repository/PrusaSlicer-settings-master/live/PrusaSlicer.version`: HTTP 0, Unsupported protocol:
Protocol "https" not supported or disabled in libcurl
```

### Solution
Use " instead of '
https://stackoverflow.com/questions/6884669/curl-1-protocol-https-not-supported-or-disabled-in-libcurl

$ curl \`https://files.prusa3d.com/wp-content/uploads/repository/PrusaSlicer-settings-master/live/PrusaSlicer.version\`
curl: (1) Protocol "`https" not supported or disabled in libcurl

$ curl "https://files.prusa3d.com/wp-content/uploads/repository/PrusaSlicer-settings-master/live/PrusaSlicer.version"
2.1.1

iksheth, 2020-03-04:
is it fixed ?

bubnikv, 2020-03-05:
It is not. It seems to be an issue of our build server since 2.2.0-alpha4. The result is that our Linux builds do not pull profile updates from our servers.

bubnikv, 2020-03-15:
fixed in PrusaSlicer 2.2.0-rc3. closing.

closed 2020-03-15 by bubnikv