prusa3d/PrusaSlicer #3553: FR - Snap every 15ºdegrees when rotating

italocjs, 2020-01-20:
### Version
2.2.0 alpha 2

### Operating system type + version
Windows 10 1903, 

### 3D printer brand / version + firmware version (if known)
Not related to printer

### Behavior (feature request)
When rotating the part is hard to get integer numbers, it doenst "snap" to integer numbers,  it would be nice to be able to rotate a part to exactly 90º using the rotate tool instead of typing it in the part property

JohnOCFII, 2020-01-20:
It will snap if you have your cursor on the white circle with the distance marks, as shown in red on this screen shot.
[image]

italocjs, 2020-01-20:
Oh thats even better!, thanks!

closed 2020-01-20 by italocjs