prusa3d/PrusaSlicer #3554: Add brim enforcer/ brim blockers

belgiancoder, 2020-01-20:
_Is this a new feature request?_
Yes

_note:_ Depending on the internal implementation of the slicer, It might be a duplicate of #2455 or even #286. But I believe my issue adds additional information, context and use case. If the community wants to close it as dupe, let's close it.

So, I have designed a double sided coat hanger. And I want the little pegs to have a brim, but not the center, which has a sufficient surface area.

[image]

So I created a part modifier only to discover that it was not possible to block brims.

[image]

As a result, my first layer looks like this.

[image]

So my feature request is to add brim blockers, just like we have support blockers.

bakman2, 2020-01-28:
I concur, would be a great feature.

BorisKozo, 2020-02-14:
I was looking for this feature as well and came to this issue :(

ThePirat91, 2020-03-24:
I am on the same boat as you.....please add this extremely useful feature on next updates!!

eivory, 2020-04-11:
A brim enforcer would be really helpful for the RC3 face shields.

stanzel, 2020-05-27:
this would be really helpful on print in place assemblies where brims in some places would cause problems

leedabiker, 2020-09-24:
I've just been looking to use this feature also. + 1 upvote from me

Photelegy, 2023-01-19:
This would be very useful.

EnduringGuerila, 2024-03-11:
Here we are 4 years later... 
please add this feature.

TheEvilRocky, 2024-03-25:
Yes, please.

Just an example. having narrow edges i would like to have also an inner brim, but do not need it on the circles.
[image]

Snuff1eupagus, 2024-05-02:
You can do this in PrusaSlicer already. 

[image]

[image]

[image]

otibsa, 2026-08-15:
So what was the solution/workaround? Those Youtube videos do not exist anymore.

BorisKozo, 2026-08-15:
@otibsa Unfortunately Prusa Slicer doesn't have the Mouse Ears tool that Bambu and Orca slicers have so you either need to manually create them using Right Click -> Add Part -> Cylinder , set it to height of 1 layer or move to the other slicers.

closed 2026-09-29 by prusaslicer-bot[bot]