prusa3d/PrusaSlicer #3557: Pause command results in infinite loop

RIGamer, 2020-01-20:
### Version
2.2.0 alpha2 win64

### Operating system type + version
Windows 10

### 3D printer brand / version + firmware version (if known)
Stock Prusa mk3s
Firmware 3.8.1
### Behavior
Using the 'Insert Pause Print (M601)' functionality results in the printer pausing at the desired level, but resuming causes it to pause again in an infinite loop

_Is this a new feature request?_No

[PauseBug.zip](https://github.com/prusa3d/PrusaSlicer/files/4088081/PauseBug.zip)

lukasmatena, 2020-01-20:
This is MK3 firmware related bug, it is already known: https://github.com/prusa3d/Prusa-Firmware/issues/1378. That's also where it should be discussed. Looking at the issue tracker and recent commits, this appears to have been already fixed by @wavexx and merged into master for [3.9.0 release](https://github.com/prusa3d/Prusa-Firmware/releases/tag/v3.9.0-beta) (currently in beta stage).

This issue does not belong here and is probably fixed, so I'll close it. If it does not work with the new FW, definitely let us know. Thanks.

closed 2020-01-20 by lukasmatena

pat0-OLI, 2020-02-18:
Worked with the 3.9.0 beta firmware release.