prusa3d/PrusaSlicer #3560: Documentation about printer configuration

photodude, 2020-01-22:
I'm looking at cleaning up a community custom PrusaSlicer profile for my BIBO 2 dual extruder. Where can I find documentation for developing [printer configurations like those included in PrusaSlicer](https://github.com/prusa3d/PrusaSlicer/tree/master/resources/profiles)
### Version
2.1.1

### Operating system type + version
win10

### 3D printer brand / version + firmware version (if known)

- Printer
  - BIBO2 Touch
    - Minor Upgrades
      - PEI sheet build surface
      - 5015 cooling blowers
- Firmware
  - manufacturer's custom version of Marlin 1.0.0 with ditto printing based on the ShaqFoo fork of Marlin 1.0.0

### Behavior
Can't find documentation for developing custom printer configurations. Looking to build [printer configurations like those included in PrusaSlicer](https://github.com/prusa3d/PrusaSlicer/tree/master/resources/profiles)

### Specific questions:
- needing direction for Dual extruder setup at the code INI level settings
- what is the meaning of the `bed_shape = -100x-100,100x-100,100x100,-100x100` definition and what do I need to change for a 214mm X by 186mm Y by 160mm Z build volume?

bubnikv, 2020-01-22:
I have posted our internal guide to PrusaSlicer's wiki so you have
something to start with.
https://github.com/prusa3d/PrusaSlicer/wiki/Vendor-bundles-and-updating-process

The wiki page needs some more love and we are extending the format of the
Preset Bundle header to support additional features of the install wizard.

You are welcome to edit the guide, the wiki should be editable to you.

Also it is likely the best for you to derive your profiles from
https://github.com/prusa3d/PrusaSlicer/tree/master/resources/profiles/Creality

> what is the meaning of the bed_shape = -100x-100,100x-100,100x100,-100x100 definition
and what do I need to change for a 214mm X by 186mm Y by 160mm Z build
volume

Those are just the four 2D coordinates corners of a square.

You may also prepare your profiles in slicer, then export them to Preset
Bundle and then edit the preset bundle to your liking, mainly adding the
header and trying to reduce config redundancy by using the inheritance.

We may ask @rtyr to help you with your further questions.

bubnikv, 2020-01-22:
Please note that the wiki page is quite raw. For example, we currently deliver our config files from our content delivery network, and we will only support downloading the configs from there for security reasons.

photodude, 2020-01-23:
@bubnikv Awesome. Thank you. I've made my first round of edits and I'm now working on reducing config redundancy. I'll need to make some other printer-specific adjustments as some things I just carried over from the suggested starting point. 

Once the redundancy is cleaned up I think it would be ready to pass to the BIBO community for verification testing.

I've attached my first draft. If you or @rtyr have any recommendations for this I would be grateful for the advice. 
[BIBO2Touch.zip](https://github.com/prusa3d/PrusaSlicer/files/4100648/BIBO2Touch.zip)

photodude, 2020-01-23:
@bubnikv @rtyr I think I have things set up correctly for dual extrusion, but with the dual extruder there are many cases where we want to only print with one of the two nozzles and keep the other extruder disabled.  With other slicers, we usually have this setup for 3 printer variants, one for Dual extruder, one for E1 right extruder only, and one for E2 left extruder only.

For the E1 only case it seems I just say that there is only one extruder, dual extrusion I indicate that there are two extruders. Where I'm lost is set up for using E2 the left extruder only, and ensuring E1 is disabled or otherwise ignored. Any guidance on that case?

rtyr, 2020-01-23:
Extruder can be selected via UI, so maybe you don't need separate printer profiles for E1/E2?

You can use T[initial_tool] placeholder in the start gcode.

I noticed that you have same extruder color for both extruders in your printer profile. I recommend to set them differently (much better preview).
[image]

bubnikv, 2020-01-24:
> For the E1 only case it seems I just say that there is only one extruder, dual extrusion I indicate that there are two extruders. Where I'm lost is set up for using E2 the left extruder only, and ensuring E1 is disabled or otherwise ignored. Any guidance on that case?

PrusaSlicer will not emit any tool change commands into the G-code for a single extruder printer. If you want to create a single extruder profile to use just the 2nd extruder, configure it as a single extruder printer and then add the selection of the 2nd extruder into your custom start G-code.

Please note that if you do that, you likely will want to add the selection of the 1st extruder into your custom start G-code for the single extruder printing profile using just the 1st extruder.

bubnikv, 2020-01-24:
I am adding a reference to https://github.com/prusa3d/PrusaSlicer/issues/3566 here, where some other issues regarding the vendor bundles configuration structure has been discussed.

photodude, 2020-02-08:
Thanks @bubnikv and @rtyr, I've taken into consideration your suggestions and made some modifications to the start gcode. I also fixed an issue with the compatible printers selection for the print profiles to ensure all the printer variants are covered.

at the moment I'll stick to the 3 printers as specialized configurations concept that the manufacturer uses on other slicers (some users get confused with how to do single extruder printing on dual extruders which is the primary justification I've heard for keeping all three variants)

One thing I'm noticing with the config bungle import is I get 2 copies of the filaments, like "generic PLA" and a second with the same name but appended with `@BIBO`. not sure why I'm getting 2 copies of all the filaments.
[BIBO2Touch.zip](https://github.com/prusa3d/PrusaSlicer/files/4173805/BIBO2Touch.zip)

I've tried to add to the Wiki as I've had questions and found answers.

bubnikv, 2020-02-08:
I don't know. I suppose some of the profiles are loaded from your filaments
directory.

Looking into your config bundle, you should add the @BIBO2 to the filament
profile names.
[filament:Generic PLA @BIBO2]

photodude, 2020-02-08:
@bubnikv Thank you, That fixed the duplicate materials.

Apparently, something after the start gcode or after the start gcode is causing prints to stop after the start gcode purge lines. It's very odd. I'm trying to get gcode from users who are reporting the issue to see if I can track down the issue. Nothing

photodude, 2020-02-08:
@bubnikv I think the stopping after start gcode issue may have been related to an extra `/n` at the end of the start gcode. not sure why a trailing `/n` would cause the printer to halt, but removing that seems to have solved the error, that or it was changing `G92 E0` to `G92 E0.0`. Checking with some other users to confirm the issue has been solved.
[BIBO2Touch.zip](https://github.com/prusa3d/PrusaSlicer/files/4175108/BIBO2Touch.zip)

At the moment I think there is one last item to check off, I've hardcoded the standby temperature for the other nozzle in the start gcode. using `M109 S170 T1` for standby when printing with just the first nozzle. it would be helpful if there was away to tie that value to a replacement like `M109 S[standby_temperature_1] T1`

I saw there was a `standby_temperature_delta` token but that's not real helpful when the start gcode doesn't allow calculations in the replacements to do something like `M109 S{first_layer_temperature_1-standby_temperature_delta] T1`

What would be the right approach for setting up standby temperatures for replacement tokens?

bubnikv, 2020-02-08:
> I saw there was a standby_temperature_delta token

Such "token" was only discussed.

> but that's not real helpful when the start gcode doesn't allow
calculations in the replacements to do something like M109
S{first_layer_temperature_1-standby_temperature_delta] T1

It does support it.

https://github.com/prusa3d/PrusaSlicer/wiki/Slic3r-Prusa-Edition-Macro-Language

What would be the right approach for setting up standby temperatures for
replacement tokens?

photodude, 2020-02-08:
@bubnikv Thank you I will look over that information. I've also added it to the Wiki.

I found the `standby_temperature_delta` token in the wiki, must be a carryover from older information and not relevant anymore. 
https://github.com/prusa3d/PrusaSlicer/wiki/Slic3r-placeholders-(a-copy-of-the-mauk.cc-page

photodude, 2020-02-08:
@bubnikv I tried the following the start gcode

`S{temperature_0 *0.8}` to attempt to get an 80% standby temperature but I got the following error.

```
G-code export to C:\Users\[user]\AppData\Local\Temp\.20796.gcode failed due to invalid custom G-code sections:
start_gcode
Please inspect the file C:\Users\[user]\AppData\Local\Temp\.20796.gcode.tmp for error messages enclosed between
!!!!! Failed to process the custom G-code template ...
and
!!!!! End of an error report for the custom G-code template ...
```

Seems either I got the syntax wrong or the macro constructs don't work right in the start gcode.

photodude, 2020-02-09:
@bubnikv Looks like I got the Syntax wrong, I found `S{first_layer_temperature[0] * 0.7}` will work but `S{temperature_0 * 0.7}` would not.

bubnikv, 2020-02-09:
The wiki on conditional G-code speaks about the difference in addressing
vector values using the old [] and new {} syntax. The new {} syntax accepts
the usual [] syntax for addressing vectors.

standby_temperature_delta is used by the "ooze prevention" feature, so the
config value is still available.

photodude, 2020-02-12:
@bubnikv Thank you. I've made some more adjustments and will be putting this to our community for more testing to make sure it's all working as expected. 
[BIBO2Touch.zip](https://github.com/prusa3d/PrusaSlicer/files/4195717/BIBO2Touch.zip)

bubnikv, 2020-02-18:
should we put your profile into our PrusaSlicer 2.2.0-beta?

photodude, 2020-02-18:
@bubnikv I'll submit a PR once I'm confident in the config bundle. It's close but not fully ready. In the meantime, I've opened a [repository in my account to track future changes with the config bundle](https://github.com/photodude/BIBO2-PrusaSlicer) while the community is testing.

I also need to figure out where to put the STL and SVG bed files to test those.

There are a couple of odd things I've noticed in my testing.

1.  With the Ooze prevention enabled the stand by temperature variation applies to the wrong extruder. It's applying to the active extruder and not to the extruder which should be in standby. In my testing with extruder T0 being first the standby temperature delta is applied to T0 and not to the T1 extruder just after the start gcode. Perhaps something needs to be added to the custom gcode to get that to apply correctly.
2.  Setting the infill to extruder 2 doesn't seem to do anything, and prints with extruder 2 set for infill have no change in the slicer preview or in the actual print.
3.  The wipe tower throws an error saying it requires relative extruder positioning. I'm unsure if the BIBO2 printer could be set to relative extruder positioning. I need to verify if the extruders can be set to relative positioning and change the profiles and start gcode to match.

photodude, 2020-03-02:
@bubnikv @rtyr Could one of you give an explanation and criteria for selecting values with the following fields? I've looked at a few things but haven't been able to wrap my head around the meaning of these fields or the criteria for selecting values for them.
```ini
parking_pos_retraction = 92
cooling_tube_length = 5
cooling_tube_retraction = 91.5
extra_loading_move = -2
```
Thank you again on the advice and guidance for building the vendor config bundle.
We are getting really close to having everything ready to push for a PR. We do have some minor issues to clean up but so far it's looking really good.

rtyr, 2020-03-02:
These settings are mainly for our multi-material upgrade or similar single-nozzle multi-material printers. It is not used for traditional dual extruder printers like BIBO2 and these settings are not even visible in the UI when the "Single Extruder Multi Material" parameter is not enabled.

filament_minimal_purge_on_wipe_tower - you can specify minimal amount of plastic to be purged on the wipe tower (if enabled). Main purpose of this setting is to make sure, that pressure in the nozzle is stabilized before printing main object.

photodude, 2020-03-02:
Thanks @rtyr that makes a lot more sense about when and where those fields apply. I've removed those fields from our config bundle, and I'll add a note to the Wiki for that. I still need to address a couple of things in the config bundle and then I'll be ready to submit a PR.

mikolaszuza, 2020-03-03:
@photodude Thanks for the edits to the Vendor bundle tutorial!

You said your profile "is close but not fully ready". Any chance it will be "good enough" for a starting profile for 2.2 final release? We would love to include profiles for at least a few different printers to kickstart the discussion about making these.

photodude, 2020-03-03:
@mikolaszuza you are welcome. Here are a few items that I need to address, I think these are the big ones holding back a PR.

- [ ] refine bed_model
- [ ] refine bed_texture
- [ ] z-seam blobs and pits
  -  needs testing following recent change to retraction settings
- [ ] reported issue with start gcode causing filament to unload
  - I haven't been able to replicate this and the solution proposed by the user reporting the issue breaks the wipe tower functionality
- [ ] reported issue with start gcode being too long for printing over USB drives causing printers to stall on start gcode
  - personally consider this one something that can't be addressed
- [ ] add soluble support print profiles
  - mostly copy from PrusaResearch.ini with minor edits

a non-blocker item would be to clean up more of the redundancy in the profiles, it's a to do item but shouldn't hold back any PRs for inclusion.

You can check the current work or participate in the development in my github repo, I have the bed_model and bed_texture in there but those are just early drafts.
https://github.com/photodude/BIBO2-PrusaSlicer

photodude, 2020-03-13:
With the adoption of the BIBO vendor profile and the creation of the Wiki document I think we have enough documentation now to close this issue. Thanks everyone for your help and guidance on this project.

closed 2020-03-13 by photodude

bubnikv, 2020-03-13:
thanks for your effort with the wiki.