prusa3d/PrusaSlicer #3561: Stepper motor not locked when changing filament

fiddlekrazy, 2020-01-22:
### Version
 2.2.0 alpha 2

### Windows 10

### ender 3
Added direct drive with no software update needed
### Upon executing the 'Change Filament Command" and parking the head, the stepper motors should be locked but are not. Any pressure applied to the head (X or Z axis) or the bed (Y axis) can cause movement
I have ruined a few prints because of this. Fortunately they were not 10 hour prints.

_Is this a new feature request?_ No? 

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_

lukasmatena, 2020-01-22:
I don't think that there is anything that we can do about it. It is up to the printer firmware how it processes the M600 gcode. If it decides that powering the steppers off is a good idea, I don't think we can override it. There is a similar issue in Cura issue tracker: https://github.com/Ultimaker/Cura/issues/5833#issuecomment-500450231 with a similar conclusion.

lukasmatena, 2020-01-22:
Now I noticed that the other issue is also authored by you, so you're probably not surprised. Have you found the "workaround in gcode"? What do you expect PrusaSlicer to do?

koenkooi, 2020-01-22:
I had this happen last week, I sent a home XY command using octoprint, that fixed it for me in that situation.

fiddlekrazy, 2020-01-22:
keonkooi: Thanks for the suggestion. 
I wrote 
M600
G28 X0 Y0
manually into the 'Add custom code'
and it worked.
This is only a workaround and not a solution but ...

bubnikv, 2020-03-15:
what to do with this issue?
Is it Marlin issue? Is it specific to Ender 3 firmware? Has it been solved by your settings?

rtyr, 2020-04-27:
This is an firmware issue. Closing.

closed 2020-04-27 by rtyr