prusa3d/PrusaSlicer #3562: Head returns to last location after doing Filament change

fiddlekrazy, 2020-01-22:
### Version
2.2.0 Alpha 2

### Operating system type + version
Window 10

### 3D printer brand / version + firmware version (if known)
Ender 3
### Behavior
When filament has been change and print is resumed, the print head returns to where it left off and deposits a blob of filament.

Instead of going to the last stop it should go to the next operation
* _Actual Results_
  * _Screenshots from __*PrusaSlicer*__ preview are preferred_

_Is this a new feature request?_ no

fiddlekrazy, 2020-01-22:
I missed some info
What happens is, after the filament change is made, it goes back to the position it was in before the change, proceeds to leave a bit of filament and then moves on to the next position. This is OK if the next position is next to it but when it is in a totally different place then the blob hardens and when the head passes that location it knocks the print askew.
From my previous post (3561) I am guessing that it is a function of 'M600' and cannot be changed through slicer software. 
I have tried to wipe the nozzle but there always seems to be a small leak.
If anyone has a workaround I would be happy to try it.

Panayiotis-git, 2020-01-23:
IIRC, there was a similar issue opened a while ago at the firmware repository. It is a not so easy to implement the move to the next position.

What I do, whenever I use the M600 is to also use a wipe tower. You can minimize per filament the required extruded amount and the size of the tower, so to minimize the losses.

yoyo42, 2020-01-26:
A wipe tower is an option I suppose. But the test extrude has just been done and confirmed by the user, then a followup short extrude is done just before the move back to the model - so I should be the case that it can jump back into printing normally. But it doesn't - I also often see a long pause when it gets back to where it left off, leaving a blob before moving on. This is variable length from nothing to several seconds, so maybe it's waiting for a reheat of something that hasn't been kept up to temp? Or waiting for some comms buffer to fill? Whatever the reason it should not pause even for a moment once it gets back into position.

rtyr, 2020-05-21:
This is about printer firmware, printer just continues to execute gcode commands after filament change sequence, which is completely controlled by the firmware.

closed 2020-05-21 by rtyr