prusa3d/PrusaSlicer #3563: Exported config bundle from 2.1.1 doesn't work OK in 2.2.0-alpha2

ochm, 2020-01-22:
### Version
PrusaSlicer-2.2.0-alpha2+win64-201912221340

### Operating system type + version
Win 10 64-CZ

### 3D printer brand / version + firmware version (if known)
MK3S

### Behavior
Exported config bundle from 2.1.1
Imported to 2.2.0-alpha2
When selected print profile other than system, almost everything from filament settings is missing, include system filament profiles. It seems only filaments based on default left in list.
If selected system print profile, all filaments are back.

My exported config bundle
[PrusaSlicer_config_bundle.zip](https://github.com/prusa3d/PrusaSlicer/files/4100011/PrusaSlicer_config_bundle.zip)

It is my bad, or some bug?

JohnOCFII, 2020-01-22:
I've successfully created a config bundle including various printer, filament and print settings profiles for a non-Prusa printer from 2.1.1 and successfully imported them into 2.2.0-alpha2 and have been using them successfully.  I'd guess something got messed up in your config bundle creation.  Mine was hand-edited to only include certain profiles.

neophyl, 2020-01-23:
This is basically the same issue as #3347

bubnikv, 2020-01-23:
I have just closed #3347, see the comment there.
Closing this one.

closed 2020-01-23 by bubnikv