prusa3d/PrusaSlicer #3564: "Export..." file chooser dialog defaults to any user-mounted drive

remuladgryta, 2020-01-23:
### Version
2.2.0-alpha2+linux-x64

### Operating system type + version
Linux Mint 18.3 Cinnamon

### 3D printer brand / version + firmware version (if known)
Original Prusa SL1

### Behavior
When a user-mounted drive is present, File->Export->Export... and File->Export->Export G-code... default to opening the file chooser dialog on one of those drives. Without an option to change this behavior, this is inconvenient if you have any internal drives mounted by a non-root user.
* _Steps needed to reproduce the problem_
  * Have any drive mounted under /media/\<username\>/.
  * Launch PrusaSlicer, import any STL.
  * Select File->Export->Export... or File->Export->Export G-code...
* _Expected Results_
  * File chooser dialog opens, defaulting to the same path as the other entries under File->Export.
* _Actual Results_
  * File chooser dialog opens, defaulting to /media/remuladgryta/my_windows_partition.

_Is this a new feature request?_
No.

kocikdav, 2020-01-23:
Thank you for reporting. We are aware that this behavior could be inconvenient to some users. Therefore there will be changes in next version. Please let us know how you like it after upcoming alpha.

bubnikv, 2020-03-15:
fixed in prusaslicer 2.2.0-rc3.

closed 2020-03-15 by bubnikv