prusa3d/PrusaSlicer #3566: Documentation on profile inheritance?

icanhazpython, 2020-01-23:
So I'm having trouble finding a concise answer on how profile inheritance works. I know you guys have an insurmountable mountain of issues here to sort through, so I'll apologize in advance and hope I don't piss off anyone with this info request. Anyways here's what I have gleaned:

- I think I may have read a post earlier from a dev here that inheritance is not *true* inheritance; it would be complicated to develop that. However I'm not sure how to reconcile that with the fact that I see chains of inheritance going sometimes 3 levels deep within the stock PrusaSlicer.ini config. I'm not really sure at this point if parameters set on say the generic PETG profile would carry over to a profile 3 times removed.
- When exporting config bundles, prusaslicer seems to like to write ALL parameters into each profile, regardless of what it inherits. Would it be advisable to strip out everything from a profile that doesn't differ from the inherited profiles? Or should I set those parameters to "nil"? I see that the new filament retraction overrides in the filament profiles implement nil for unset parameters, however I'm not seeing nil used elsewhere in the profiles. 
- I'd like to have a config bundle with my custom profiles, but I'm divided on whether I should keep the export as-is, with prusaslicer adding every possible parameter, or if I should strip out anything that hasn't changed from its inherited profiles, in the interest of keeping things clean, maintainable, and retaining the ability to benefit from any updates to the inherited profiles that come in from you guys. I guess I'm wondering what kind of updates generally happen to pre-existing profiles, and would I want to allow those updates to potentially alter my custom profiles, or should I keep my profiles completely detached from any automated updates to the in-built profiles? I'm looking for a general consensus here. 

I hope that this thread could benefit others looking to distribute custom profiles in a clean and maintainable manner. Thanks in advance.

bubnikv, 2020-01-23:
> I think I may have read a post earlier from a dev here that inheritance is not true inheritance

It is a true inheritance.

> I'm not really sure at this point if parameters set on say the generic PETG profile would carry over to a profile 3 times removed.

They do.

> When exporting config bundles, prusaslicer seems to like to write ALL parameters into each profile

There is a good reason for that. Such file contains all the information. If we change the system profiles, you will see those parameters marked as differing from the system preferences.

> Would it be advisable to strip out everything from a profile that doesn't differ from the inherited profiles? 

yes

> Or should I set those parameters to "nil"? 

no

> I'd like to have a config bundle with my custom profiles, but I'm divided on whether I should keep the export as-is, with prusaslicer adding every possible parameter, or if I should strip out [...]

If you are creating your profiles for your machine, you should not inherit from PrusaResearch configs. Indeed, once you provide us with your config bundle and if it is structured to be used by the install wizard, then PrusaSlicer does not allow inheriting across these config bundles. You really should follow the example of the Creality profile that we are providing with the latest PrusaSlicer-2.2.0-alpha2.

icanhazpython, 2020-01-23:
Thanks for the reply. I'm looking at the creality profile now. A couple questions regarding that:

- Is there any technical significance to the "@" appended to the config name: `print:0.12mm DETAIL @ENDER3`? Or just a way to delineate in the UI that said config is intended for a specific printer? Any reason to do this even when the profiles will be shown/hidden based on the compatible printers condition?
- When you said "provide us with your config bundle and if it is structured to be used by the install wizard, then PrusaSlicer does not allow inheriting across these config bundles", I'm surmising you mean that anything calling for inheritance in Creality.ini will not be able to inherit from PrusaResearch.ini, so in effect each of these configs is completely siloed off from each other?

The specific configs that I'm interested in sharing are only filament configs. I used to have printer profiles for retraction settings, but with the new retraction overrides, I've been able to pull those settings into the filament profiles and get rid of my printer profiles. So thanks for that! My aim is to distribute some filament profiles on github for a few filaments that I like to print with and have tweaked for best results on the prusa series of printers.

rtyr, 2020-01-23:
The "@" sign in the preset name is a separator of a logical name from the full name. Logical name of the print preset is used in the main screen. It is also used in the configuration wizard (filaments selection). See release notes for 2.2.0-alpha1,

Inheriting across different config bundles is not allowed (each bundle is using its own filament profiles).
[image]

icanhazpython, 2020-01-23:
Ah I get it! I think all my questions have been answered. Thanks for that little lesson, I appreciate it and I'm sure others will as well when they go looking for this thread. 👍

closed 2020-01-23 by icanhazpython

bubnikv, 2020-01-24:
> Is there any technical significance to the "@" appended to the config
name: print:0.12mm DETAIL @ENDER3? Or just a way to delineate in the UI
that said config is intended for a specific printer? Any reason to do this
even when the profiles will be shown/hidden based on the compatible
printers condition?

That the profile names have to be unique across vendors and across machine
types. The text after @ ensures the profile name is unique.

bubnikv, 2020-01-24:
>  Ah I get it! I think all my questions have been answered. Thanks for
that little lesson, I appreciate it and I'm sure others will as well when
they go looking for this thread

We would appreciate, if the community could help with the documentation in
return :-) I believe you can edit the wiki.

bubnikv, 2020-01-24:
> When you said "provide us with your config bundle and if it is structured
to be used by the install wizard, then PrusaSlicer does not allow
inheriting across these config bundles", I'm surmising you mean that
anything calling for inheritance in Creality.ini will not be able to
inherit from PrusaResearch.ini, so in effect each of these configs is
completely siloed off from each other?

Exactly. We have to find a balance between maintainability, complexity of
the concept, code complexity and data duplication.

We decided to go with the simple to understand and implement approach with
the penalty of some data duplicities.

We internally discussed whether it is worth to have one "global"
configuration, from which the others could derive, but then we decided to
postpone it if the need for such a "mother" profile emerges. In this case
it is easier to add this functionality later than to remove something once
it is introduced. The "global" configuration may be used for filaments, but
then an FDM printer prints at its best if the filament profiles are fine
tuned to it, so sharing the filament profiles does not make much sense, so
we decided to bind the filament profiles across printer vendors just by
"logical names" aka "aliases". It is a simple to grasp concept and we
believe it will make the life to contributors easier than the tightly
coupled system implemented by Cura. We found out that it is quite a
challenge to design printer profiles for our printers to be integrated into
Cura indeed due to its tight coupling with filaments data and complexity of
the configuration structure.