prusa3d/PrusaSlicer #3567: FR - Catch 3MF file name instead of body1904810

italocjs, 2020-01-23:
### Version
2.2.0 Alpha 2

### Operating system type + version
Not related

### 3D printer brand / version + firmware version (if known)
Not related

### Behavior
It woud be nice to have "catch file name" when importing a 3MF with only one body instead of naming the part as body 184018 (or some random number that i have no idea how its generated and doenst really makes sense, there is one body, why the heck would it be called 31930).

This "catching" the filename already happens well with STL (in single body, and if separated multiple bodies it will then assign a _1, _2 ... to the end

This is not so much of a issue, but its pretty annoying and i have no idea on how to change the pslicer program by myself

bubnikv, 2020-01-23:
Sorry, but I cannot understand what you mean.

italocjs, 2020-01-23:
Expected:
Original 3MF file name:  readable_name.3MF
Imported model name: readable_name
Exported Gcode name: readable_name..gcode

What happens:
Original 3MF file name:  readable_name.3MF
Imported model name: Body_381831038
Exported Gcode name: Body_381831038.gcode

I have no idea where the numbers ( in example 381831038 ) comes from,  this only happens to 3MF, not to STL, i have recently started using 3MF after seeing a video from josef about its advantages

FidelCapo, 2020-01-24:
Can you please attach your 3MF and STL files? Thank you

bubnikv, 2020-02-19:
The problem is not clear. No response, closing.

closed 2020-02-19 by bubnikv

italocjs, 2020-02-19:
i have no idea how clearer i could be, but no problem, i just stopped using 3mf's