prusa3d/PrusaSlicer #3568: unerwarteter Filamentwechsel

Martin1161, 2020-01-23:
### Version
_Version of PrusaSlicer used goes here_
Slicer Version: PrusaSlicer-2.1.0+win64-201909160915 basiert auf Slic3r
_Use `About->About PrusaSlicer` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #s_
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_

### 3D printer brand / version + firmware version (if known)
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_

WINDOWS 10, RAM 8GB, Prozessor INTEL i5-6402PCPU 2,80GHz, 64-Bit Betriebssystem x64 basierter Prozessor

### Behavior ---> see below
* _Describe the problem_
* _Steps needed to reproduce the problem_
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*PrusaSlicer*__ preview are preferred_

_Is this a new feature request?_

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_

Hallo zusammen,

Slicer Version: PrusaSlicer-2.1.0+win64-201909160915 basiert auf Slic3r
Drucke: Original Prusa i3 MK3

Druckeinstellung 0,2 mm Quality MK3 - geändert auf 4 Konturen und Bienenwabe
Filament PETG

Erwartetes Verhalten:
unterbrechungsloser Druck, ohne Filamentwechsel

Problembeschreibung:
Druck unterbricht ständig, nachdem die ersten 3 mm problemlos gedruckt, wurden  den Weiterdruck und fordert einen Filamentwechsel auf. Dies geschieht während des druckes einer Schicht ca 3mal. Ca. 20 Wechsel (Filament herausgezogen und wieder eingelegt) wurden durchgeführt, danach wurde der Druck manuell gestoppt. Druck besteht aus drei einzelnen STL Objekten (filamentbedarf berechnet ca 110gr Druckzeit erwartet 11h58 min.

Ich hoffe ihr könnt weiterhelfen. 
LG Martin

config.ini:
# generated by PrusaSlicer 2.1.0+win64 on 2020-01-23 at 23:26:39 UTC
avoid_crossing_perimeters = 0
bed_custom_model = 
bed_custom_texture = 
bed_shape = 0x0,250x0,250x210,0x210
bed_temperature = 85
before_layer_gcode = ;BEFORE_LAYER_CHANGE\nG92 E0.0\n;[layer_z]\n\n
between_objects_gcode = 
bottom_fill_pattern = rectilinear
bottom_solid_layers = 4
bridge_acceleration = 1000
bridge_angle = 0
bridge_fan_speed = 50
bridge_flow_ratio = 0.95
bridge_speed = 30
brim_width = 5
clip_multipart_objects = 1
colorprint_heights = 
compatible_printers_condition_cummulative = "printer_notes=~/.*PRINTER_VENDOR_PRUSA3D.*/ and printer_notes=~/.*PRINTER_MODEL_MK3.*/ and nozzle_diameter[0]==0.4";"nozzle_diameter[0]!=0.6 and ! (printer_notes=~/.*PRINTER_VENDOR_PRUSA3D.*/ and printer_notes=~/.*PRINTER_MODEL_MK(2.5|3).*/ and single_extruder_multi_material)"
complete_objects = 0
cooling = 1
cooling_tube_length = 5
[... 191 lines cut ...]
toolchange_gcode = 
top_fill_pattern = hilbertcurve
top_infill_extrusion_width = 0.4
top_solid_infill_speed = 40
top_solid_layers = 5
travel_speed = 180
use_firmware_retraction = 0
use_relative_e_distances = 1
use_volumetric_e = 0
variable_layer_height = 1
wipe = 1
wipe_into_infill = 0
wipe_into_objects = 0
wipe_tower = 1
wipe_tower_bridging = 10
wipe_tower_rotation_angle = 0
wipe_tower_width = 60
wipe_tower_x = 170
wipe_tower_y = 125
wiping_volumes_extruders = 70,70
wiping_volumes_matrix = 0
xy_size_compensation = 0
z_offset = 0

[Druckproblem.pptx](https://github.com/prusa3d/PrusaSlicer/files/4106153/Druckproblem.pptx)

rtyr, 2020-01-24:
It looks like issue with the filament sensor. Filament sensor can be turned off (lcd menu), you can try it.

bubnikv, 2020-03-20:
no response, closing.

closed 2020-03-20 by bubnikv