prusa3d/PrusaSlicer #3569: Thin walls cause overlapping gapfill and infill in same area

vintagepc, 2020-01-24:
### Version
2.2.0-alpha2

### Operating system type + version
OpenSuSE 42.3, x64

### 3D printer brand / version + firmware version (if known)
Mk3S/MMU2

### Behavior
Sliced a part with 2mm walls and was reviewing the layers. I noticed it's trying to put both gap fill and regular infill in between the walls, which will lead to overextrusion, jamming, etc.

Turning off 'Detect thin walls' removes the gap fill, so it is correctly recognizing that normal infill in this area would cause overextrusion, but the problem is it does not appear to be disabling the regular infill when adding the gap fill.

 Picture:
[image]

_Is this a new feature request?_ No

#### Project File (.3MF) where problem occurs
[speaker_Test_body.3mf.gz](https://github.com/prusa3d/PrusaSlicer/files/4106659/speaker_Test_body.3mf.gz)

bubnikv, 2020-03-20:
This issue may have been fixed with a9fc39491e0b37f9a68ac99c372f09121e4c8afd

bubnikv, 2020-03-20:
There was certainly a bug in PrusaSlicer related to perimeter generator, which was sometimes putting extrusions into wrong region "bins" due to the logic of merging regions of same perimeter settings but different infill parameters. This bug should be fixed with a9fc394 and the fix will be part of the 2.2.0 final.

vintagepc, 2020-03-20:
Great - that does indeed sound like a viable cause and fix. I'll keep an eye out to see if it recurs.

rtyr, 2020-03-30:
Fixed in the 2.2.0 final. I was unable to reproduce this issue with the provided 3MF. Closing.

closed 2020-03-30 by rtyr