prusa3d/PrusaSlicer #3570: 2.2.0-A3: Sequential Printing: Wrong location after slicing

foreachthing, 2020-01-24:
### Version
2.2.0-alpha3; 
a0e6afcee871d74de2d7dacb1e798215e62646ca

### Operating system type + version
Windows 10 x64

### 3D printer brand / version + firmware version (if known)
UM2

### Behavior
* Adding multiple Parts to the plate
* Change settings for sequential printing (print settings -> output -> Complete individual parts [check]; settings for my UM2: R 60, H 50 mm)
* Click that annoying message about "objects too close" about a 1000 times ;-) (once would suffice; or click arrange automatically, if CIP is active)
* Arrange parts
* SLICE now

Plate:
[image]

Sliced:
[image]

Observe: the whole plate has been moved +XY, even that far, that it would print outside the bed.

#### Project File (.3MF) where problem occurs
https://www.thingiverse.com/thing:566112
[Battery_Adapter_AA-C.zip](https://github.com/prusa3d/PrusaSlicer/files/4107399/Battery_Adapter_AA-C.zip)

bubnikv, 2020-01-24:
Thanks, there was a refactoring being done on sequential print. I will look
into this issue.

BTW, wouldn't you interested in providing system profiles for your
Ultimaker 2 to PrusaSlicer?
You may find the following two issues and a wiki helpful
https://github.com/prusa3d/PrusaSlicer/wiki/Vendor-bundles-and-updating-process

https://github.com/prusa3d/PrusaSlicer/issues/3566
https://github.com/prusa3d/PrusaSlicer/issues/3560

foreachthing, 2020-01-24:
> BTW, wouldn't you interested in providing system profiles for your Ultimaker 2 to PrusaSlicer?

Well, yeah, sure (_after looking into your PrusaResearch.ini, this might take some time, though_)!
I'll start my own repo or can I PR to yours somewhere?

bubnikv, 2020-01-24:
> I'll start my own repo or can I PR to yours somewhere?

We plan to create a github "organization", where each vendor would have a separate repo created. As of now, a pull request into PrusaSlicer github is appropriate.

bubnikv, 2020-01-24:
Thanks for reporting the bug, it is a regression from 71fa41110049a79ec1d5c3052d22647a38d01bed.

bubnikv, 2020-01-24:
Fixed with dced3ce783c9a3a1f626bf9497fa5d611146f3bf
Thanks for help.
Closing

closed 2020-01-24 by bubnikv