prusa3d/PrusaSlicer #3571: Black bars around icons in Gtk3

hyperair, 2020-01-24:
### Version

2.1.1

### Operating system type + version

Ubuntu 19.10

### Description

Originally reported at https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=949333, there are black bars around the print/filament/printer settings icons on the main screen in the Debian package, but not in the pre-compiled tarball.

I'm not sure what causes these exactly, but I do notice that the pre-compiled tarball distributed in Github is compiled against gtk2, while the Debian package is compiled against gtk3.

tamasmeszaros, 2020-01-24:
Hello @hyperair,
PrusaSlicer with GTK3 based wxWidgets is something we haven't payed much attention in the recent releases. It would be best to link against GTK2 until we polish up GTK3 build. It is hard to say when is this going to happen but if Debian Sid is wants to get rid of GTK2, we might as well hurry up the transition.

hyperair, 2020-01-26:
@tamasmeszaros Debian no longer has a libwx compiled against gtk2, so gtk3 is our only choice.

Salamandar, 2020-04-07:
I'm reporting the same issue on Archlinux, with wxWidgets 3.0. It seems to disappear with wxWidgets 3.1, although new issues appear (reported in https://github.com/prusa3d/PrusaSlicer/issues/4014)

fiechr, 2020-04-09:
Came to confirm the same issue on Arch, when building from source.

Some more screenshots: https://imgur.com/a/pzcN4za

When using the static build provided as binary or AppImage, the glitches disappear (also the ones mentioned in #4014).

volca02, 2020-11-03:
The fix from this comment works to remedy this on Arch with wx 3.0: https://github.com/prusa3d/PrusaSlicer/issues/1016#issuecomment-419345674

bubnikv, 2020-12-12:
Merged, thanks all involved.

closed 2020-12-12 by bubnikv