prusa3d/PrusaSlicer #3575: Problem Stützstruktur

mcn1, 2020-01-26:
Hallo,
ich habe ein Problem mit einem Modell.

PrusaSlicer richtet nur am Rand eine Stützstruktur ein!
Dieses Modell müsste aber komplett gestützt werden.

Ich habe schon alles probiert. Haben Sie eine Lösung?

PS: Auf einer Ecke würde sogar eine Stützstrucktur in die Luft gedruckt!?

mfg mcn1

Datei:
[Wandhalterung_FritzFon_C6.zip](https://github.com/prusa3d/PrusaSlicer/files/4113403/Wandhalterung_FritzFon_C6.zip)
Config:
[config.zip](https://github.com/prusa3d/PrusaSlicer/files/4113408/config.zip)

Modell:
[image]
Vorschau:
[image]
Stützstruktur:
[image]
[image]
Einstellungen:
[image]
Version:
[image]

Stützstrucktur in der Ecke:
[image]

### Version
_Version of PrusaSlicer used goes here_

_Use `About->About PrusaSlicer` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #s_
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_

### 3D printer brand / version + firmware version (if known)
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_

### Behavior
* _Describe the problem_
* _Steps needed to reproduce the problem_
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*PrusaSlicer*__ preview are preferred_

_Is this a new feature request?_

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_

foreachthing, 2020-01-26:
Bitte das nächste Mal auf Englisch - Danke!

Ich habe mir Dein Model heruntergeladen und mit der neusten Version (https://github.com/prusa3d/PrusaSlicer/releases) ausprobiert (jedoch ohne Deine Config). Alles wie es sollte ... !

[image]

[image]

Schnapp Dir also die neuste Version und versuche es nochmals.

Short: get the latest version of PS and try again.

BerndJM, 2020-01-26:
1. bei "Brücken nicht unterstützen" das Häkchen raus, sonst wird die große Fläche als als Brücke druckbar interpretiert und dem entsprechend nicht unterstützt.
2. "Stützmaterieal nur auf dem Druckbett" sollte bei dem Modell völlig ausreichen

mcn1, 2020-01-26:
Hello,

thank you foreaththing for the fast answer!
I downloaded the newest Version, installed it and slice it with startsettings.
But it is was the same problem!

Thank you BerndJM!
Your tip with parameter "Don't support bridges" was the problem. 
Without this parameter i became what i want.

mfg mcn1

rtyr, 2020-03-28:
I believe we can close this. I don't speak German very well, but it seems that your problem was solved by disabling "Don't support bridges" parameter.

closed 2020-03-28 by rtyr