prusa3d/PrusaSlicer #3577: "Add Colour Change" should pick random colours.

LaserBee, 2020-01-26:
In the latest version, when I slide to a layer and hit "ADD COLOUR CHANGE", a colour picker comes up, I have to pick a colour, and hit OK. 

Could we have an option to turn that feature off, and just have it so when I hit "ADD COLOUR CHANGE" it just picks one at random without asking for confirmation? 

I've found the colour picker to be of no use to me, and just add an extra couple mouse-clicks to my workflow. I'm sure some people like it, but I'd much rather be able to switch it off!

Thanks for all your hard work.

YuSanka, 2020-01-27:
This issue is already in our ToDo list and its implementation probably will be a part of next release. Close as a duplicate of #3449

closed 2020-01-27 by YuSanka

LaserBee, 2020-01-27:
YuSanka, thank you very much. It was me who raised the issue last time, I just didn't know if anyone had even seen it. I'm sorry, I'm very new to Github, so I don't really know how this works. 

Thank you for implementing it! I really appreciate it, and all that's being done!