prusa3d/PrusaSlicer #3578: Scaling cuts off model in preview and print

lunatic1972, 2020-01-27:
version_2.2.0-alpha1-353-gdced3ce78
Hash version: dced3ce783c9a3a1f626bf9497fa5d611146f3bf
Operating System: Linux (Manjaro)
Steps to reproduce issue:
1. Load a model (I used https://www.thingiverse.com/thing:1040192)
2. I wanted to scale model to be at least 85mm high. Changed size (Z) to 85mm which then scaled X and Y proportionally
3. 3D editor view, model changes accordingly
4. Press SLICE NOW. In the Preview pane you will notice the model is cutoff at 69.60 (May depend on your slicing settings)
See picture for details
[image]

JohnnyDeer, 2020-01-29:
Hi, can you please upload 3mf file? To be able to reproduce the issue. Thank you.

lunatic1972, 2020-01-30:
[OniSkull.zip](https://github.com/prusa3d/PrusaSlicer/files/4134980/OniSkull.zip)
This one is the original one not scaled to the dimensions in the issue above.

foreachthing, 2020-01-30:
Workaround: use the "variable layer height" just on the tips of the horns. Just a "small" click will suffice :-)

bubnikv, 2020-01-31:
please try to recompile against the current master. I believe this issue
has been fixed with  c5fa400208d46f9382ba992a7796d80ae90f0609 yesterday.

lunatic1972, 2020-01-31:
I have not printed it again however in the preview it appears that it is fixed. Great job on that. I'm sure a lot of people will be thanking you for that, including myself :)

lukasmatena, 2020-01-31:
Great, I'm closing the issue then.

closed 2020-01-31 by lukasmatena