prusa3d/PrusaSlicer #3579: Adding Support enforcers/blockers or adding part/modifiers wrong z location in position

lunatic1972, 2020-01-27:
version_2.2.0-alpha1-353-gdced3ce78
Hash version: dced3ce
Operating System: Linux (Manjaro)
Steps to reproduce issue:
1. Load any model
2. Add a support blocker (or any in the subject title)
3. Location of Z axis will be flush with the plate however the Z value will be negative
4. Entering in a Z value will become an offset of the original value that was in there
See pictures

Just want to say that whoever is helping with the project, great job. Keep it up. Thanks for all the hard work
[image]
[image]

bubnikv, 2020-02-06:
The Z coordinate of the modifier is referenced to the center of the object. This is the behavior since PrusaSlicer 2.0. It may be counter intuitive and we may change that one day.

smurfix, 2020-04-06:
Well … I don't understand how that is not considered to be a bug, because the whole thing is anything but intuitive. Consider this sequence, from #3985 :
* I add a box. It has Z=0 and is placed on the bed by definition.
* I add another box on top, with a 2mm gap. It has Z=27 which seems to refer to the build plate, a.k.a. to the first object's bottom. No (obvious) center-based calculation here.
* I add an intermediate layer (a third box, but with height 2mm) and want to position it between these two. With Z=25 it floats somewhere in the center of the second box.
* Things get worse when I shrink the first box. The second box does move down, but only half way, i.e. the gap widens.

I need to break out the pocket calculator to figure out which Z values to use here. IMHO this is the kind of job I have a computer for …

closed 2026-09-29 by prusaslicer-bot[bot]