prusa3d/PrusaSlicer #3580: Make reloading Models easier

akloster, 2020-01-27:
In Cura, you can reload models by pressing F5, and it will also detect changes to the file automatically and ask you if you want to reload the changed file.

PrusaSlicer can only reload models through a context menu at the "edit" icon in the models table. It took me a while to even find it there.

I would suggest a way to reload models globally (like with F5) and an option to track file system changes.

In Cura, I mostly used this feature with models generated from scripts, so it would also be nice if a script/process could ask Slicer to reload a file and reslice. That would be more complicated, of course.

bubnikv, 2020-01-30:
this proposal certainly makes sense, especially if the behavior will match
the muscle memory of Cura users, Windows users etc.

bubnikv, 2020-02-19:
[image]

bubnikv, 2020-02-19:
Implemeted. F5 is available through the Edit menu and at the Plater and object list.
The change will be part of the upcoming PrusaSlicer 2.2.0-beta. Closing.

closed 2020-02-19 by bubnikv