prusa3d/PrusaSlicer #3582: Wipe/Prime Tower with different purge volumes makes the left over grid in the wrong side

joaomamede, 2020-01-28:
### Version
PrusaSlicer-2.2.0-alpha2+linux-x64-201912221342

### Operating system type + version
Linux 64bits appimage
No 3D rendering issues

### 3D printer brand / version + firmware version (if known)
Ender 3 + Dual Extruder/switching extruder + Marlin 2.0

### Behavior
When creating a prime tower with different purge volumes, say Black to Golden to be 120mm³ and Golden to Black to be 75mm³ for example.

When going from black to golden it uses the whole length to do the 120mm³ just like in this picture.
[image]

When going from Golden to black it starts from a part of the inside of the tower (golden side) and goes all the way to the black portion of the tower, and then to complete the difference between the purge volumes, it goes to the golden purge part to make the walls that will be bridges (when going black to golden). Just like in this picture.
[image]

When this happens and when it's purging to golden, the tip touches black PLA and the first bit of portions of the golden parts get shaded black (or any other stronger color).

I have two suggestions that could fix this. 
- When doing the completion rectangles maybe they should always be done prior to unloading of the golden filament. 
- Another option, is to make the opposite of what is happening.
Start in the middle of the length of the tower and always do the squares on the right size (the one that requires less purging, in this example, black part).

* _Steps needed to reproduce the problem_
Just add a dual extrusion project.

rtyr, 2021-04-28:
If this issue is still valid, please provide your 3MF project and the layer where do you think is some issue. I don't see any issue on your example pictures, the yellow part is ramming sequence before the unload of the gold filament. Gold filament will not be used on the same layer again.

bubnikv, 2021-06-03:
No response, closing.

closed 2021-06-03 by bubnikv