prusa3d/PrusaSlicer #3583: Cooling Fan problem with tool change

J-charles-C, 2020-01-28:
### Version
_Version of PrusaSlicer : 2.1.1+

### Operating system type + version
Mac OSX 10-14-6

### 3D printer brand / version + firmware version (if known)
Personnel Hypercube (coreXY) with Marlin 2.0.1 

### Behavior
I,
I use three extruders with a single nozzle. When I print an object with several extruders, the cooling fan only works with T0. It is at 0 with T1. I do not have this problem with CURA. In my profile, I have two filaments, configured exactly the same with cooling fan on.

Best

[Archive.zip](https://github.com/prusa3d/PrusaSlicer/files/4122410/Archive.zip)

J-charles-C, 2020-01-31:
I,

Why configuration label ?

best

lukasmatena, 2020-01-31:
I didn't find anything wrong with the fan. The `M106` and `M107` commands seem to be where they should be. Could you be more explicit about where the error in the gcode is?

There are definitely issues with your custom g-codes, though. At least `G92E0` should be `G92 E0`. However, I doubt it could fool your printer enough to turn fan off.

brettvitaz, 2020-01-31:
I have noticed this problem as well. In my printer (Ender 3, Marlin 2.0), tool 1 had stored fan and temp settings in the printer. Switching from tool 0 to tool 1 would automatically execute the stored settings, which caused issues when switching tools while printing. I had to add fan speed and temperature commands to my tool change script.

J-charles-C, 2020-01-31:
I,

Indeed, after the tool change, if I restart the fan with the printer screen, it stays on during the following tool changes.

best

lukasmatena, 2020-01-31:
>  It's a Marlin Bug ?

Maybe the intended behaviour at some point, I don't know. I would look here: https://github.com/MarlinFirmware/Marlin/issues/15877

Zemistr, 2020-02-08:
I have right now the same problem and honestly, I really don't want to have external script converting 
```
M106 S([0-9]{1-3})
```
to
```
M106 P0 S$0
M106 P1 S$0
```

I think the slicer has to count with fan speed for each extruder and during tool change just set the proper temperature.

Or maybe new macro [current_fan_speed]?

https://www.facebook.com/groups/1049718498464482/permalink/2151042491665405/

VanessaE, 2020-03-16:
Still an issue in 2.2.0-rc3.  Prusaslicer is applying fan speeds in a way incompatible with how Marlin handles tool changes, and there is no way to make a custom toolchange gcode workaround, as neither the layer number nor the current fan speed are available there.

Zemistr, 2020-03-16:
@VanessaE Here is post-process script. It takes your file, finds all M106 in it and replace them to set M106 for two extruders in the same time. ;)

[GCodeFixFans.zip](https://github.com/prusa3d/PrusaSlicer/files/4337392/GCodeFixFans.zip)

closed 2026-09-29 by prusaslicer-bot[bot]