prusa3d/PrusaSlicer #3585: Always Sequential Printing Error (Regardless of setting, placement or object size)

ellisgoward, 2020-01-29:
### Version
_2.2.0-Alpha-2_

### Operating system type + version
_Windows 10 Pro 64Bit_

### 3D printer brand / version + firmware version (if known)
_Creality Ender 3 Pro_

### Behavior
* I'm getting the too close error. Regardless of object placement or clearance distance.
* I have tried all sorts of clearance distances in the settings.
* I've tried placing object at opposite ends of the bed.
* This was working well for me on 2.0 & seems to be an Issue with the latest Alpha releases.

Many Thanks,
Ell

[image]
[image]
[image]

vintagepc, 2020-01-30:
IIRC there's a known ordering "gotcha" here - it does not reconfigure the order it prints objects in from the front of the bed to the back (the objects are printed in the order they are added, I think - gets even more interesting if you use the arrange function and it scrambles them all over the bed...), so in many cases it will want to print the back object first and then fail because you'd end up swinging the extruder tail/X axis over top of it to print the objects closer to the front. In the interim you may have success if you swap the object places or rotate them to be side-by-side rather than front-to-back

bubnikv, 2020-02-03:
could you please provide your 3mf?

closed 2020-02-07 by bubnikv

bubnikv, 2020-02-07:
The validation algorithm did not take object rotation into account by mistake. Fixed with 348bf39
The fix will be part of the PrusaSlicer 2.2.0-alpha4.
Closing.

commit b8c898bf (2020-02-07): Fixed a bug in validation of the FDM Print to check for extruder
collisions against already printed objects: Rotation of an object
was not being taken into account.
Fixes #2450 PrusaSlicer 2.0 Sequential printing vs rotate object
Fixes #3585 Always Sequential Printing Error (Regardless of setting, placement or object size)

Fixed some collisions in sequential print just after arrangement by
making the validation using a slightly lower extruder radius value
than the arrangement.

Refactored PrintObject coordinate system so that the PrintObject's
coordinate system is always centered in respect t [... cut]