prusa3d/PrusaSlicer #3587: Prusa Slicer doesnt start

JackGelee2, 2020-01-30:
Prusa Slicer 2,1,1 win 64

Hello,

the actual version prusaslicer doesn t start. The programm open an shows evertying without the printer window (3D Window with the printerplate)

My window version ist Windows 10 64 pro.

JackGelee2, 2020-01-30:
[image]

enricoturri1966, 2020-02-03:
Duplicate of https://github.com/prusa3d/PrusaSlicer/issues/2939 ?

JackGelee2, 2020-02-03:
Yes is was the same problem. I try to solve it. Now i can start the program and read the stl file. When i start the slicing process the program freeze.
The problem is the graphic card with the software (newest nvidia driver gtx 1080). If i uninstall the graphic card I can use the slicer. I try an old version (1 year old) graphic driver and its the same.
Windows 10 64 pro with all updates doesn t work with nvidia and the prusa slicer :-(

Sandeant, 2020-02-04:
Hi dears,

I had the same issue with my new PC on Ryzen9 and Radeon X5700XT and I found how to fix it - I just enabled SVM Mode (Virtualization) in bios for my CPU, it was disabled by default.

tamasmeszaros, 2020-02-06:
Hello,

I've tried to disable SVM mode on a Ryzen 7 3700 with Radeon RX 560 and PrusaSlicer (2.1.1 and latest master also) works without problems. It seems unlikely to me that this would cause a direct problem. Though, it may disrupt the graphics driver of RX5700, hard to say for sure...

Sandeant, 2020-02-09:
Yes, it can be specific of my config - AMD Ryzen9 3950x +  MB Asus ROG x570 (last bios 1405) + Radeon RX5700XT. On this configuration without SVM, no version of PusaSlicer worked for me, even the oldest like Slic3r1.3

tamasmeszaros, 2020-03-16:
Ok @Sandeant, if PrusaSlicer now runs on your configuration, I'll close this issue. Good luck!

closed 2020-03-16 by tamasmeszaros