prusa3d/PrusaSlicer #3588: Program immediately exits after loading, a window never appearing

nebbywan, 2020-02-01:
### Version
Commit be92e109857b68348b3682e53f7a16ad0a099f75

### Operating system type + version
Win10 x64 1909 build 18363.628

### Behavior
When compiled and installed as "release" prusa-slicer.exe can be seen in Task Manager loading and unloading in under a second. When attempting launch a debug build from VS2019 the following output is displayed as it immediately exits
[Output.txt](https://github.com/prusa3d/PrusaSlicer/files/4142343/Output.txt)

I rebuilt deps, just in case, but this had no effect. The official alpha2 release does not exhibit this behavior for me. In case it helps, here is the build output:
[Build.txt](https://github.com/prusa3d/PrusaSlicer/files/4142346/Build.txt)

tamasmeszaros, 2020-02-06:
Hi!

please, make sure that `libgmp-10.dll` and `libmpfr-4.dll` are placed beside the `PrusaSlicer.dll`. The CMake build system should ensure this after a successful build, but it might not worked in your case.
If they are in fact missing, you can find them in the build tree in `deps/GMP/lib/` and `deps/MPFR/lib/` folders respectively.

foreachthing, 2020-02-06:
And it also did not work in my case.
I, too, had to copy those files manually.

tamasmeszaros, 2020-02-06:
Ok, I will look at the CMake script, to find out what the issue is.

tamasmeszaros, 2020-02-06:
@foreachthing, @nebbywan please confirm if the provided fix works for you.

foreachthing, 2020-02-06:
Sweet!
It's all there! Thanks a bunch!
[image]

tamasmeszaros, 2020-02-06:
Ok, I think we can close and move on :)

closed 2020-02-06 by tamasmeszaros

nebbywan, 2020-02-07:
Commit 2e039d3955a5f8d47abdd46e9a0c505b1f477dd6 fixed the issue, thanks @tamasmeszaros!