prusa3d/PrusaSlicer #3589: Asymmetric SLA support

d01010101, 2020-02-01:
### Version
2.2.0-alpha2

### Operating system type + version
Ubuntu 18

### Behavior
See that heavily asymmetric support. Possibly it makes sense with Prusa, but how would it make sense for a generic SLA printer?

Is it a bug or a specific adaptation to Prusa? How to obtain a symmetric support for a generic SLA printer?
[image]

d01010101, 2020-02-01:
By the way, a variant of this model, where the smaller element was missing and the larger was rotated by 90 deg, required an enormous amount of memory to produce the support, which was also  asymmetric. I do not have the other model anymore and I do not want to recreate that situation, because my 8GB system heavily swapped and it took it several minutes to return to a normal state.

tamasmeszaros, 2020-02-06:
Hi!

I think this is fixed in the latest master branch, so it will be out with the next release.

closed 2020-03-16 by tamasmeszaros