prusa3d/PrusaSlicer #3590: Feature Request: tickbox to wait for bridge fan ramp up time

madsi1m, 2020-02-01:
### Version
2.2.0-alpha2

### Operating system type + version
unimportant

### 3D printer brand / version + firmware version (if known)
Marlin 2 from github

### Behavior
In my Marlin config i have set `#define FAN_KICKSTART_TIME 1000` which will pulse my parts cooling fan for 1000ms (1s) at 100% to get it moving then will set the speed to the correct value. 

What I have noticed, especially in bridging, is the fan is not at the correct speed when the bridge starts to print, it first goes to 100% for FAN_KICKSTART_TIME miliseconds, then reduces to what i have actually set as "bridges fan speed" in PrusaSlicer.

### Feature Request
Can we have a tickbox with a settable wait time that pauses before bridges start to print?

For example: with the box ticked and set to 500ms, the gcode before printing the beginning of a bridge would:
1. maybe do a retraction?
2. set the fan speed to "bridges fan speed"
3. wait 500ms
4. start printing the bridge.

The wait should only happen when the setting is selected and once before a bridge cycle.

Pizzahd88, 2020-10-02:
I think this could be a valuable feature, but think the implementation could be done in a better manner.

My recommendation would be to have the option of giving the slicer a bridge-fan-delay option, where you could put in ex --500ms. The slicer would then set the fan speed = bridge fan speed while travelling to the bridge.
At the time the nozzle arrives at the bridge the fans should be around the speed wanted.

mozartlovescats, 2023-02-01:
This is a very useful feature that I think may be very critical for some users. When printing high temperature filaments it's almost guaranteed to have the first part of the bridge sagging down because the fan is off by default. If the fan is already spooling up then the bridge will come out perfect.

Standing still waiting is not optimal, we can spool up the fan while printing the last 1s (time to be defined as parameter) in the path, then the fan is already spooling up and can continue with bridge.

I'm very surprised that this hasn't been implemented yet.

unameisfine, 2023-05-21:
Merge with #10004 please!

rcmaniac25, 2024-07-10:
I see this as a filament override too, as some filaments may need it more then the print, or inversely, some filaments may have problems. So an override would be good.

VooDisss, 2025-03-05:
4-5 years passed by... Sadly this is still not implemented (whereas in superslicer it is, but it has no tree supports now) :(

madsi1m, 2025-03-05:
try OrcaSlicer

synthead, 2025-10-01:
I am also very interested in this feature

Tupson444, 2026-09-15:
still relevant